tabemann/zeptoforth #2: Block editing on Discovery STM32F746

sbobyrne, 2021-10-02:
I have uploaded version 0.20.2 (the _full binary) on a Discovery STM32F746G board.  
I am connecting via a USB serial interface.
I can connect to forth using either e4thcom or picocom.  However, when I try to use the block editor, on e4thcom it does not edit the block at all but only gives text to the right of the box drawn on the screen. I have to use \ enter to exit the terminal and then reboot the device.
On picocom, I can use the line editor, but backspace does not work and when I do C-W, C-V to exit, and type 1 load, the interpreter tells me it can't find the block.  I am obviously missing something but don't know what.  The C-W does seem to save as the asterisk at the head of the block goes away.

Terminal commands
e4thcom -t noforth -b B115200 -d ttyACM3
picocom -b 115200 /dev/ttyACM3 --imap lfcrlf --omap delbs,crlf --send-cmd "ascii-xfr -s -n"

I'm trying to use the block system as an internal workspace for developing code, and am hoping they will save to flash.  Let me know if that's not the intention.

It looks like a nice forth, and I'm very keen on having a block editor, which is why I'm trying it out.  But I'm struggling with trying to work out how to make the editor work, it seems.

tabemann, 2021-11-21:
Sorry for missing your issue! I have been rather preoccupied lately with getting zeptoforth working on the RP2040, and have not been checking on Github lately! By the way, neither block editor nor the line editor will work on e4thcom, and e4thcom renders the line editor largely superfluous. For picocom, I have not tried that terminal program, and I wouldn't be surprised if there were terminal issues (I have only tried those out under swdcom, by the way). As for 1 load, are you sure that the block you are using isn't block 0, because IIRC zeptoforth starts blocks from index 0.

tabemann, 2021-11-21:
I just checked the man page for picocom, and the problem is that zeptoforth assumes delete is DEL, not BS, which is why delete is not working for you.

frenchie68, 2022-06-10:
Side note: block numbers always start at 0. However, you are not allowed to LOAD block 0 because in standard compliant implementations, it means that you are operating from the console.

tabemann, 2022-06-10:
zeptoforth does not pretend to be standards-compliant, even if much of the time it does approximate the standards, and I should note, and `0 LOAD` means to load block 0 in it (because it has no privileged "block" 0 for the console).