tabemann/zeptoforth #5: STATE in not maintained consistently

frenchie68, 2022-06-10:
Hi Travis,

I have been somehow challenge to prove the value of your output. I believe it is promising but you are not quite there yet. You seem to adhere, in a way, to the ANS-94 standard specification but you're not there yet.  I tried to run some OTS benchmarks on your platform on the STM32F11RE. Although it is not officially supported, this test pattern ought to work yet it does not:

tabemann, 2022-06-10:
I should note that `."` does not work when `STATE` is 0, which is per standard. I also should warn you that while zeptoforth is pretty close to ANS, it does not make a concerted effort to follow ANS in every way (e.g. it uses `NOT` rather than `INVERT` because `INVERT` is an abomination, and it uses `[IMMEDIATE]` and `[COMPILE-ONLY]` within words rather than `IMMEDIATE` and `COMPILE-ONLY` after words due to issues imposed by compiling to flash).

tabemann, 2022-06-10:
If you want to print a constant string at the REPL, use `.(`, which of course is closed with `)`, to print it, because `.(` is immediate.

tabemann, 2022-06-10:
I should note that while zeptoforth does support `S"` and `C"` at the REPL, these are highly non-standard, and impose their own restrictions (i.e. because it stores the strings in temporary buffers that eventually get reused for more temporary strings).

tabemann, 2022-06-11:
I have made a new release, 0.40.0, which adds `."` and `.\"` at the REPL level, so one is no longer required to be in a compilation state to use them (in full builds at least - in kernel-only and mini builds compilation state is still required).

frenchie68, 2022-06-11:
Thanks Travis. /MOD (on the F411) and .S now look good to me. I haven't run extensive tests but as far as I can tell, they are good to go. Do you implement floored division? That's also a big thing since FORTH-83.