tabemann/zeptoforth #8: Can't install ZeptoForth on RP2040 Pico

tigadad, 2022-08-05:
I have an RP2040 Pico board running MicroPython. I am trying to install ZeptoForth.

I have followed the instructions which are basically the same as for installing MicroPython. I copy the .uf2 file to the Pico. It shows copying the file and then goes blank (as usual). Reboot the Pico and select the environment (bottom right) and no sign of ZeptoForth.

I am using Thonny.

What have I missed please.

ctrilsbach, 2022-08-05:
I have had this problem before intermittently when copying the file from my Raspberry Pi to the Pico. The file copies but the drive does not unmount at the end. If I do the same thing from my PC it works fine everytime. It appears to be a hardware issue and not anything to do with Zeptoforth.

tabemann, 2022-08-07:
> I have an RP2040 Pico board running MicroPython. I am trying to install ZeptoForth.
>
> I have followed the instructions which are basically the same as for installing MicroPython. I copy the .uf2 file [...]

I have never used Thonny myself, but I am under the impression that MicroPython supports USB virtual COM ports built in, whereas zeptoforth only supports serial and swdcom, so one will likely need to use a USB-serial dongle if one wishes to talk to a board with zeptoforth installed over USB.

Note that I highly recommend using [zeptocom.js](https://tabemann.github.io/zeptocomjs/zeptocom.html) under Chrome, Chromium, Edge, or Opera (it does not work with Firefox, and i have heard of some having issues with using it with FreeBSD) for talking with zeptoforth over serial. It is specifically designed for working with embedded Forth and zeptoforth in particular.

tabemann, 2022-08-07:
> I have had this problem before intermittently when copying the file from my Raspberry Pi to the Pico. The file copies but the drive does not unmount at the end. If I do the same thing from my PC it works fine everytime. It appears to be a hardware issue and not anything to do with Zeptoforth.

I am not sure what I can do to help here aside from recommending that one manually dismount the Pico if it does not manually unmount itself.