tabemann/zeptoforth #16: Issue Monterey MacOS 12.x, most recent releasw.

RandoSY, 2023-01-23:
Builds fine, loads OK exempt for complaining about unmounted improperly, bit I can;\'t see new usb serial device in /dev?

RandoSY, 2023-01-23:
Seems to build OK?

`(base) admin@admins-Mac-mini zeptoforth % make device=rp2040
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C src/stm32f407
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../common kernel_info
rm -f kernel_info.s
echo '        define_word "kernel-platform", visible_flag' > kernel_info.s
echo '_kernel_platform:' >> kernel_info.s
echo '        push {lr}' >> kernel_info.s
echo '        string "'stm32f407'"' >> kernel_info.s
echo '        pop {pc}' >> kernel_info.s
echo '        end_inlined' >> kernel_info.s
echo >> kernel_info.s
echo '        define_word "kernel-version", visible_flag' >> kernel_info.s
echo '_kernel_version:' >> kernel_info.s
echo '        push {lr}' >> kernel_info.s
echo '        string "'0.59.4-dev'"' >> kernel_info.s
echo '        pop {pc}' >> kernel_info.s
echo '        end_inlined' >> kernel_info.s
echo >> kernel_info.s
echo '        define_word "kernel-date", visible_flag' >> kernel_info.s
echo '_kernel_date:' >> kernel_info.s
echo '        push {lr}' >> kernel_info.s
echo '        string "'Mon Jan 23 15:07:11 EST 2023'"' >> kernel_info.s
echo '        pop {pc}' >> kernel_info.s
echo '        end_inlined' >> kernel_info.s
echo >> kernel_info.s
mkdir -p ../../obj
arm-none-eabi-as -g -o ../../obj/zeptoforth.stm32f407.o zeptoforth.s
arm-none-eabi-ld ../../obj/zeptoforth.stm32f407.o -T zeptoforth.ld --cref -Map ../../zeptoforth.stm32f407.map -o ../../zeptoforth.stm32f407.elf
arm-none-eabi-objdump -D ../../zeptoforth.stm32f407.elf > ../../zeptoforth.stm32f407.list
arm-none-eabi-objcopy ../../zeptoforth.stm32f407.elf ../../zeptoforth.stm32f407.bin -O binary
arm-none-eabi-objcopy ../../zeptoforth.stm32f407.elf ../../zeptoforth.stm32f407.ihex -O ihex
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C src/stm32f411
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../common kernel_info
rm -f kernel_info.s
echo '        define_word "kernel-platform", visible_flag' > kernel_info.s
echo '_kernel_platform:' >> kernel_info.s
echo '        push {lr}' >> kernel_info.s
echo '        string "'stm32f411'"' >> kernel_info.s
echo '        pop {pc}' >> kernel_info.s
echo '        end_inlined' >> kernel_info.s
echo >> kernel_info.s
echo '        define_word "kernel-version", visible_flag' >> kernel_info.s
echo '_kernel_version:' >> kernel_info.s
echo '        push {lr}' >> kernel_info.s
echo '        string "'0.59.4-dev'"' >> kernel_info.s
echo '        pop {pc}' >> kernel_info.s
echo '        end_inlined' >> kernel_info.s
echo >> kernel_info.s
echo '        define_word "kernel-date", visible_flag' >> kernel_info.s
echo '_kernel_date:' >> kernel_info.s
echo '        push {lr}' >> kernel_info.s
echo '        string "'Mon Jan 23 15:07:11 EST 2023'"' >> kernel_info.s
echo '        pop {pc}' >> kernel_info.s
echo '        end_inlined' >> kernel_info.s
echo >> kernel_info.s
mkdir -p ../../obj
arm-none-eabi-as -g -o ../../obj/zeptoforth.stm32f411.o zeptoforth.s
arm-none-eabi-ld ../../obj/zeptoforth.stm32f411.o -T zeptoforth.ld --cref -Map ../../zeptoforth.stm32f411.map -o ../../zeptoforth.stm32f411.elf
arm-none-eabi-objdump -D ../../zeptoforth.stm32f411.elf > ../../zeptoforth.stm32f411.list
arm-none-eabi-objcopy ../../zeptoforth.stm32f411.elf ../../zeptoforth.stm32f411.bin -O binary
arm-none-eabi-objcopy ../../zeptoforth.stm32f411.elf ../../zeptoforth.stm32f411.ihex -O ihex
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C src/stm32l476
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../common kernel_info
rm -f kernel_info.s
echo '        define_word "kernel-platform", visible_flag' > kernel_info.s
echo '_kernel_platform:' >> kernel_info.s
echo '        push {lr}' >> kernel_info.s
echo '        string "'stm32l476'"' >> kernel_info.s
[... 66 lines cut ...]
echo '        string "'0.59.4-dev'"' >> kernel_info.s
echo '        pop {pc}' >> kernel_info.s
echo '        end_inlined' >> kernel_info.s
echo >> kernel_info.s
echo '        define_word "kernel-date", visible_flag' >> kernel_info.s
echo '_kernel_date:' >> kernel_info.s
echo '        push {lr}' >> kernel_info.s
echo '        string "'Mon Jan 23 15:07:12 EST 2023'"' >> kernel_info.s
echo '        pop {pc}' >> kernel_info.s
echo '        end_inlined' >> kernel_info.s
echo >> kernel_info.s
mkdir -p ../../obj
arm-none-eabi-as -g -o ../../obj/zeptoforth.rp2040.o zeptoforth.s
arm-none-eabi-ld ../../obj/zeptoforth.rp2040.o -T zeptoforth.ld --cref -Map ../../zeptoforth.rp2040.map -o ../../zeptoforth.rp2040.elf
arm-none-eabi-objdump -D ../../zeptoforth.rp2040.elf > ../../zeptoforth.rp2040.list
arm-none-eabi-objcopy ../../zeptoforth.rp2040.elf ../../zeptoforth.rp2040.bin -O binary
arm-none-eabi-objcopy ../../zeptoforth.rp2040.elf ../../zeptoforth.rp2040.ihex -O ihex
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C boot
arm-none-eabi-as -g -o ../../../obj/boot.rp2040.o boot.s
arm-none-eabi-ld ../../../obj/boot.rp2040.o -T boot.ld --cref -Map ../../../boot.rp2040.map -o ../../../boot.rp2040.elf
arm-none-eabi-objdump -D ../../../boot.rp2040.elf > ../../../boot.rp2040.list
arm-none-eabi-objcopy ../../../boot.rp2040.elf ../../../boot.rp2040.bin -O binary
arm-none-eabi-objcopy ../../../boot.rp2040.elf ../../../boot.rp2040.ihex -O ihex
./boot_with_checksum.py ../../../boot.rp2040.bin ../../../boot_with_checksum.rp2040.bin
./make_uf2.py ../../boot_with_checksum.rp2040.bin ../../`

RandoSY, 2023-01-23:
Window 11 machine doesn't see it either ... wondering where *.uf2 binary is?

RandoSY, 2023-01-23:
https://github.com/tabemann/zeptoforth/releases

tabemann, 2023-01-23:
The UF2 file is in the release tarball at zeptoforth-VERSION/bin/VERSION/rp2040/zeptoforth_full-VERSION.uf2 where VERSION is the version of the release you downloaded. Once you use the USB Mass Storage device to upload the UF2 file, use a USB serial device connected to pins 1 (GPIO 0, UART0 TX) and 2 (GPIO 1, UART0 RX) and use a terminal emulator (I recommend zeptocom.js, if you have access to Chrome or Chromium; as a web terminal it is online at https://tabemann.github.io/zeptocomjs/zeptocom.html) to connect to the resulting USB CDC device.

As for building zeptoforth by yourself, by default it puts the 
UF2 image at zeptoforth.rp2040.uf2 at the base directory path. However I do not recommend this approach unless you plan on creating a new release on top of it as that is just the zeptoforth kernel alone, which is not very functional by itself.

RandoSY, 2023-01-23:
OK thanks, I got it. I need to use an additional USB2TLL converter to talk serial, I naively assumed that a usb instance would be created as it is in Micro and Circuit python. :)

RandoSY, 2023-01-23:
I as able to apply this easily to the  Cytron Maker Pi: rp2040 ...

[image]

Can  you point me to a simple Getting Started document ... ???

[image]

RandoSY, 2023-01-23:
This the zeptoforth_full-0.59.3.uf2 image from today.

[image]

tabemann, 2023-01-23:
@RandoSY I should note that zeptoforth is meant to be used with a terminal that supports ANSI colors, such as zeptocom.js, or picocom on top of a typical terminal program such as Terminal.app. That is why you were seeing such ugly garbage, because the terminal program you were using clearly does not support ANSI colors.

RandoSY, 2023-01-24:
Yes, I see that, really neat project  ... shows a tremendous amount of thought.

RandoSY, 2023-01-25:
There is a very nice blend of the past and future, block editor on one hand, all kinds of support for multiprocessing/tasking, and message passing on the other. Be happy to share what I learn, I have a lot of various rp2040 hardware as they are not expensive, and still easy to get.