tabemann/zeptoforth #18: Zeotoforth on the

hinsech475, 2023-02-02:
I recently read on github that: Patch-level release 0.59.3.1
@[tabemann](https://github.com/tabemann) tabemann released this 3 days ago
· [4 commits ](https://github.com/tabemann/zeptoforth/compare/v0.59.3.1...master)to master since this release
[ v0.59.3.1](https://github.com/tabemann/zeptoforth/tree/v0.59.3.1)
https://github.com/tabemann/zeptoforth/commit/1a78b0402eedd40c91c2ca87d9248b94df4d5334
This patch-level release adds official support for the SeeedStudio XIAO RP2040, adds a Neopixel driver for the RP2040, adds the ability to turn off console colors, and fixes a stack garbage issue.

tabemann, 2023-02-04:
You gave me the idea to buy three WIO RP2040 mini Dev Boards in hopes that I manage to not break one of them when soldering them (I'm terrible at killing boards while soldering). I note from your posts on Forth2020 that you got zeptoforth running on one of your WIO RP2040's. Once I get my boards I will try to get the WiFi radio on it working with zeptoforth (it does not support it as of yet, but does have SPI support that can be used with it).

arlaneenalra, 2023-05-22:
OK!

One failure mode:

I have an oled hooked up to gpio 14/15 and have all of the files from the wiki Hello World example loaded [oled.fs.txt](https://github.com/tabemann/zeptoforth/files/11537170/oled.fs.txt). I concatenated them in order and set them up to compile-to-flash ... but that was just a convenience. 

Running:

Note: I have the draw-string arguments out of order there .. with `op-set` in the wrong position.
```
oo import
bitmap import
ssd1306 import
font import
simple-font import

128 constant my-width
64 constant my-height

my-width my-height bitmap-buf-size constant my-buf-size
my-buf-size 4 align buffer: my-buf
<ssd1306> class-size buffer: my-ssd1306
14 15 my-buf my-width my-height SSD1306_I2C_ADDR 1 <ssd1306> my-ssd1306 init-object

init-simple-font

s" Hello, World!" my-width 2 / 7 13 * 2 / - my-height 2 / 8 2 / - my-ssd1306 op-set a-simple-font draw-string
$FF my-width 4 / my-width 2 / 0 my-height op-xor my-ssd1306 draw-rect-const
my-ssd1306 update-display
```

Several times in succession yields the Following:
[failure-usb.txt](https://github.com/tabemann/zeptoforth/files/11537185/failure-usb.txt)

and on the serial console it yields:

[failure-serial.txt](https://github.com/tabemann/zeptoforth/files/11537189/failure-serial.txt)

The last run crashed the chip so it no longer responds on usb. Sometimes it's failing with:

```
\ Put your Forth code to upload here.  ok
\   ok
\ Clicking 'Send' without a selection will upload the contents of this area to the target.  ok
\   ok
\ Clicking 'Send' with a selection will upload just the selection to the target.  ok
  ok
oo import wordlist order overflow
Error
```
on the usb console ... I wonder if there's a memory/stack exhaustion issue with error cases some where .. or the console is falling back to serial and dropping usb .. I just got it to reboot from the serial side through a debug probe even though it was using a serial console. Also, I'm not seeing stack traces on the USB console at all, they only show up on the serial side ... 

Could that be what's going on? Something crashes out the USB stack and it's falling back to serial entirely?

arlaneenalra, 2023-05-22:
It could also be a red herring .. the situation I ran into this with might just not be easily recoverable ... though it is annoying you can't get stack traces on the USB side ..
hmm

tabemann, 2023-05-22:
@arlaneenalra The wordlist order overflow message is because each time you ran your code, you pushed more wordlists onto the current wordlist order. Sooner or later it will overflow. A good way around this is to use the following:

```
continue-module forth
  oo import
  <put your code here>
end-module
```

That way your code will be in the default wordlist but will pop `oo` from the wordlist order. Note, however, if there is an error in between, you may have to manually execute `oo unimport` to remove it from the wordlist order.

arlaneenalra, 2023-05-22:
> @arlaneenalra <https://github.com/arlaneenalra> About the stack traces on
> the serial console only, that is because they are generated with the bare
> minimum functionality, as they assume that the system is in an undefined
> state, and USB is just too complex for them (i.e. too likely to not work
> when the system is in an undefined state).

That makes sense and probably classifies as "do dumb stuff win dumb prizes"
kind of error ...

tabemann, 2023-05-23:
@arlaneenalra I should note that

```
s" Hello, World!" my-width 2 / 7 13 * 2 / - my-height 2 / 8 2 / - my-ssd1306 op-set a-simple-font draw-string
```

is practically guaranteed to fail spectacularly; `op-set` is 0, btw. I'm almost surprised that you were able to recover at all from that.