tabemann/zeptoforth #41: Minor issue in the wiki

arlaneenalra, 2023-05-22:
https://github.com/tabemann/zeptoforth/wiki/Bitmaps,-Fonts,-and-I2C-SSD1306-based-Displays-for-zeptoforth#hello-world

The code listed to write "Hello World!" uses `set-string` instead of `draw-string`. In particular the line that looks like:

`s" Hello, World!" my-width 2 / 7 13 * 2 / - my-height 2 / 8 2 / - my-ssd1306 a-simple-font set-string`

should instead look something like:

`s" Hello, World!" my-width 2 / 7 13 * 2 / - my-height 2 / 8 2 / - op-set my-ssd1306 a-simple-font draw-string`

I'm not overly familiar with github wiki's or I'd offer a more direct fix ..

tabemann, 2023-05-22:
Thanks for the heads-up! I have now fixed the example as you mention and tested it.

arlaneenalra, 2023-05-22:
Looks good to me!

closed 2023-05-22 by arlaneenalra