tabemann/zeptoforth #44: Does the documentation mention number parsing?

robzed, 2023-06-17:
I think Zeptoforth supports direct hex numbers $4A3 , decimals #123 , binary %10011 , doubles 12.3 and fixed point 12,4 ... but I don't see these documented anywhere?

I looked at basic Git front page, wiki or supplied release package docs. 

Of course, people could just look at the source code :-)

tabemann, 2023-06-17:
I have added more content on numbers to [the wiki](https://github.com/tabemann/zeptoforth/wiki/Getting-Started-with-zeptoforth). In the process of this I have discovered two things - first, I don't have parsing routines exposed to the user to parse double-cell and fixed-point numbers, and there's a major bug in parsing numbers with bases smaller than 2 or larger than 16.

tabemann, 2023-06-18:
I have now implemented `parse-double`, `parse-double-unsigned`, and `parse-fixed` to enable parsing double-cell integers and fixed-point numbers from strings, and have fixed the bug in the handling of unsupported bases.

tabemann, 2023-06-18:
Note that this is not in a build yet but is in the `master` and  `devel` branches.

robzed, 2023-06-18:
Fantastic!

closed 2023-12-26 by tabemann