tabemann/zeptoforth #53: Error parsing address, when uploading rp2040_full.fs

rdoetjes, 2023-08-02:
1) I cloned the reppository.

2) I ran make

3) Uploaded the uf2 using 

4) From the zeptoforth source directory, I ran  ../e4thcom-0.8.5.2/e4thcom -t noforth -b B115200 -d ttyUSB0

5) Then typed 
#include src/rp2040/forth/setup_full.fs 

ERROR that occured:

 \ Iterate executing an xt over a byte array
  : citer ( ??? addr count xt -- ??? ) ( xt: ??? c -- ??? )
    { xt } over + { addr end-addr }
    begin addr end-addr u< while addr c@ xt execute 1 +to addr repeat **unable to parse: addr**
at Line 70 in File ./src/common/forth/lambda.fs

rdoetjes, 2023-08-02:
I reran using codeload3.py 

command:
zeptoforth$ ./utils/codeload3.py -B 115200 serial src/rp2040/forth/setup_full.fs 

TX: \ Dump 16 bytes of ASCII
TX: : dump-ascii-16 ( c-addr u -- )
TX: [char] | emit
TX: dup 16 min 0 ?do
TX: over i + c@ dup $20 >= over $7F < and if emit else drop [char] . emit then
error over i + c@ dup $20 >= over $7F < and if emit else drop [char] . **emit then loop variable not found**

Error file src/rp2040/forth/setup_full.fs line 25: could not upload file
>>>  #include src/common/forth/basic.fs

Then I get another error, seems to be upload related. Is there away to bundle the forth into the uf2 perhaps?

rdoetjes, 2023-08-02:
I reflashed it completely and ran codeload3.py and that worked. 
../e4thcom-0.8.5.2/e4thcom should not be used to upload, it's not reliable

closed 2023-08-02 by rdoetjes

tabemann, 2023-08-02:
@rdoetjes It has been quite a while since I have used e4thcom, but the matter is that all the setup*.fs files expect a clean system, with only a kernel on it, or else thy will fail in one fashion or another. Note that if you have an already-installed system `erase-all` will achieve this; there is no need to reflash the kernel.