tabemann/zeptoforth #55: Are there problems with NVIC_ISPR_SETPEND@ ??

robzed, 2023-08-10:
```
interrupt import  ok
0 NVIC_ISPR_SETPEND@ . 
*** HARD FAULT *** 

Exception state: 
    IPSR:           00000003
    XPSR:           21000200
    Return address: 20002DF6 bit@ 20002DF2 +00000004
    LR:             1001A5C5 NVIC_ISPR_SETPEND@ 1001A580 +00000044
    SP:             20041FE0
    R12:            4001801C
    R11:            00000000
    R10:            FFFFFFFF
    R9:             FFFFFFFF
    R8:             FFFFFFFF
    R7:             20041BFC
    R6:             00000001
    R5:             00000001
    R4:             00000000
    R3:             00000000
    R2:             00000000
    R1:             20002DF3 bit@ 20002DF2 +00000000
    R0:             00018000

Data stack:
    TOS:      00000001
    20041BFC: 200153A4

Return stack:
    20041FEC: 20004143 main 2000412E +00000014
    20041FF0: 200076EB try 200076B8 +00000032
    20041FF4: 20041FF4
    20041FF8: 20041BFC
    20041FFC: 20004011 quit 20003FFE +00000012

Returning main task to prompt
NVIC_ISPR_Base hex u. decimal E000E200  ok
NVIC_ISPR_Base @ hex u. decimal 18000  ok
NVIC_ISPR_Base 4 + @ hex u. decimal 0  ok
NVIC_ISPR_Base 8 + @ hex u. decimal 0  ok
```

Does this mean all bits are in one word rather than separate bits in different words? 

```
  \ Get NVIC interrupt set-pending
  : NVIC_ISPR_SETPEND@ ( u -- bit )
    dup 32 / cells NVIC_ISPR_Base + @ swap 32 mod 1 swap lshift swap bit@
  ;

```

robzed, 2023-08-10:
This is the the RP2040 ...

tabemann, 2023-08-10:
Oh, this is a very obvious bug that went uncaught for so long because I never bother to check interrupts' registers for anything, but rather only set them. I have just made a fix in the `devel` branch, which I have merged into the `cyw43_driver` (really everything WiFi and IP-related) branch, which I will be testing shortly.

tabemann, 2023-08-10:
It appears to now work, from some cursory testing; I have merged my fix into the `master`  branch  now.

robzed, 2023-08-11:
> uncaught for so long because I never bother to check interrupts' registers for anything,

I also have been struggling with finding why my GPIO interrupts are not working - so added some words that support IRQ visualisation. Maybe this will be helpful? 

https://github.com/robzed/ukmarsbot_forth/blob/master/for_rp2040/irq_info.fth

NOTICE: This is a work in progress and likely has bugs!