tabemann/zeptoforth #58: RP2040 GPIO interrupt enable locations can't be accessed as bytes

robzed, 2023-08-11:
See **bold** items

Assuming 
25 constant ENC_LEFT_CLK   \ Nano D2
15 constant ENC_RIGHT_CLK  \ Nano D3
16 constant ENC_LEFT_B     \ Nano D4
17 constant ENC_RIGHT_B    \ Nano D5

irq_status 
                                        1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 
        0 1 2 3 4 5 6 7 8 9 A B C D E F 0 1 2 3 4 5 6 7 8 9 A B C D E F 
pending 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
enabled 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 _ _ _ _ _ _ _
NMI proc0 00000000
NMI proc1 00000000

1=Low Level, 2=High Level, 4=Edge Low, 8=Edge High
INTR07-00 111111EE
INTR15-08 D1222111
INTR23-16 1111115A
INTR31-24 00111151
INTE07-00 00000000
INTE15-08 00000000
INTE23-16 00000000
**INTE31-24 00000000**
INTS07-00 00000000
INTS15-08 00000000
INTS23-16 00000000
INTS31-24 00000000
 ok
**true ENC_LEFT_CLK PROC0_INTE_GPIO_EDGE_HIGH!  ok**
irq_status 
                                        1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 
        0 1 2 3 4 5 6 7 8 9 A B C D E F 0 1 2 3 4 5 6 7 8 9 A B C D E F 
pending 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
enabled 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 _ _ _ _ _ _ _
NMI proc0 00000000
NMI proc1 00000000

1=Low Level, 2=High Level, 4=Edge Low, 8=Edge High
INTR07-00 111111ED
INTR15-08 D1222111
INTR23-16 1111115A
INTR31-24 00111151
INTE07-00 00000000
INTE15-08 00000000
INTE23-16 00000000
**INTE31-24 00808080**
INTS07-00 00000000
INTS15-08 00000000
INTS23-16 00000000
INTS31-24 00000000
 ok
**false ENC_LEFT_CLK PROC0_INTE_GPIO_EDGE_HIGH!  ok**
irq_status 
                                        1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 
        0 1 2 3 4 5 6 7 8 9 A B C D E F 0 1 2 3 4 5 6 7 8 9 A B C D E F 
pending 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
enabled 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 _ _ _ _ _ _ _
NMI proc0 00000000
NMI proc1 00000000

1=Low Level, 2=High Level, 4=Edge Low, 8=Edge High
INTR07-00 111111EE
INTR15-08 D1222111
INTR23-16 1111115A
INTR31-24 00111151
INTE07-00 00000000
INTE15-08 00000000
INTE23-16 00000000
[... 376 lines cut ...]
0x13c PROC1_INTE3 Interrupt Enable for proc1
0x140 PROC1_INTF0 Interrupt Force for proc1
0x144 PROC1_INTF1 Interrupt Force for proc1
0x148 PROC1_INTF2 Interrupt Force for proc1
0x14c PROC1_INTF3 Interrupt Force for proc1
0x150 PROC1_INTS0 Interrupt status after masking & forcing for proc1
0x154 PROC1_INTS1 Interrupt status after masking & forcing for proc1
0x158 PROC1_INTS2 Interrupt status after masking & forcing for proc1
0x15c PROC1_INTS3 Interrupt status after masking & forcing for proc1
0x160 DORMANT_WAKE_INTE0 Interrupt Enable for dormant_wake
0x164 DORMANT_WAKE_INTE1 Interrupt Enable for dormant_wake
0x168 DORMANT_WAKE_INTE2 Interrupt Enable for dormant_wake
0x16c DORMANT_WAKE_INTE3 Interrupt Enable for dormant_wake
0x170 DORMANT_WAKE_INTF0 Interrupt Force for dormant_wake
0x174 DORMANT_WAKE_INTF1 Interrupt Force for dormant_wake
0x178 DORMANT_WAKE_INTF2 Interrupt Force for dormant_wake
0x17c DORMANT_WAKE_INTF3 Interrupt Force for dormant_wake
0x180 DORMANT_WAKE_INTS0 Interrupt status after masking & forcing for dormant_wake
0x184 DORMANT_WAKE_INTS1 Interrupt status after masking & forcing for dormant_wake
0x188 DORMANT_WAKE_INTS2 Interrupt status after masking & forcing for dormant_wake
0x18c DORMANT_WAKE_INTS3 Interrupt status after masking & forcing for dormant_wake

These words are located in: https://github.com/tabemann/zeptoforth/blob/24afbed0aefceb06f72e8a519cf5cc456fc52a5e/src/rp2040/forth/gpio.fs

Tested with https://github.com/robzed/ukmarsbot_forth/blob/master/for_rp2040/irq_info.fth

robzed, 2023-08-11:
Here are another four potential replacement routines (please remove underscores before using).
_**Consider these untested.**_ 

  \ Clear a raw edge low interrupt
  : _INTR_GPIO_EDGE_LOW! ( index -- )
    dup 7 and 2 lshift 2 + bit swap 1 rshift $c and INTR0 + !
  ;

  \ Clear a raw edge high interrupt
  : _INTR_GPIO_EDGE_HIGH! ( index -- )
    dup 7 and 2 lshift 3 + bit swap 1 rshift $c and INTR0 + !
  ;

  \ Get an edge low interrupt status for processor 0
  : _PROC0_INTS_GPIO_EDGE_LOW@ ( index -- enable )
    dup 7 and 2 lshift 2 + bit swap 1 rshift $c and PROC0_INTS0 + bit@
  ;

  \ Get an edge high interrupt status for processor 0
  : _PROC0_INTS_GPIO_EDGE_HIGH@ ( index -- enable )
    dup 7 and 2 lshift 3 + bit swap 1 rshift $c and PROC0_INTS0 + bit@
  ;

robzed, 2023-08-11:
I checked for the byte addressability of CPU memory mapped I/O registers.
The only thing I found was: "Control and status registers within the DW_apb_ssi are byte-addressable."

The GPIO addressing in C is int32 addressed, but I didn't check all locations. 
e.g. https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/rp2_common/hardware_gpio/gpio.c

https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_structs/include/hardware/structs/iobank0.h

tabemann, 2023-08-11:
Thank you for all this work! I myself had thought that these registers were byte-addressable up until now.

tabemann, 2023-08-12:
I have now updated the `gpio` module and done some testing of this, and it appears to work. The code is now in the `master`, `devel`, and `cyw43_driver` branches.

tabemann, 2023-08-12:
I am closing this issue as it appears to be resolved.

closed 2023-08-12 by tabemann