tabemann/zeptoforth #64: Does task-free exist?

robzed, 2023-09-14:
Listed in zeptoforth-1.0.3.1/docs/words/basic.html

Does this exist?

tabemann, 2023-09-16:
There is no task-free functionality, but one can create task pools and reuse tasks created with them once those tasks terminate. There is `kill`, which simply kills a task, but this is very liable to leave the system in an undefined state. What is recommended if one wants to terminate a task cleanly is `signal`, which will raise an exception within the task in question, which will flow downwards so all the exception-handling code can execute. However, with `signal`, there is no guarantee that the exception will actually be re-raised by any handler, and thus one cannot be sure that the task will terminate.

robzed, 2023-09-16:
You should remove it from the documentation.

tabemann, 2023-09-16:
Oh I feel like an idiot right now. I thought you were asking if there was a means to free tasks, not whether the word `task-free` itself exists. And no, the word `task-free` does not exist because I renamed it from `task-free` to `task-unused` at the same time I updated `free` to `unused` (so `free` could be used for the heap allocator) but forgot to update `basic.md` accordingly. I have now updated the docs in git reflecting this change. Thanks for mentioning this!

robzed, 2023-09-18:
Happy to close.

closed 2023-09-18 by tabemann