tabemann/zeptoforth #72: Stops compiling

robzed, 2023-10-27:
Send this from Zeptocom.js edit window

create animated_jsw
hex
3c00 h, 3c00 h, 7e00 h, 3400 h, 3e00 h, 3c00 h, 1800 h, 3c00 h, 7e00 h, 7e00 h, f700 h, fb00 h, 3c00 h, 7600 h, 6e00 h, 7700 h, 
0f00 h, 0f00 h, 1f80 h, 0d00 h, 0f80 h, 0f00 h, 0600 h, 0f00 h, 1b80 h, 1b80 h, 1b80 h, 1d80 h, 0f00 h, 0600 h, 0600 h, 0700 h, 
03c0 h, 03c0 h, 07e0 h, 0340 h, 03e0 h, 03c0 h, 0180 h, 03c0 h, 07e0 h, 07e0 h, 0f70 h, 0fb0 h, 03c0 h, 0760 h, 06e0 h, 0770 h, 
00f0 h, 00f0 h, 01f8 h, 00d0 h, 00f8 h, 00f0 h, 0060 h, 00f0 h, 01f8 h, 03fc h, 07fe h, 06f6 h, 00f8 h, 01da h, 030e h, 038c h, 
0f00 h, 0f00 h, 1f80 h, 0b00 h, 1f00 h, 0f00 h, 0600 h, 0f00 h, 1f80 h, 3fc0 h, 7fe0 h, 6f60 h, 1f00 h, 5b80 h, 70c0 h, 31c0 h, 
03c0 h, 03c0 h, 07e0 h, 02c0 h, 07c0 h, 03c0 h, 0180 h, 03c0 h, 07e0 h, 07e0 h, 0ef0 h, 0df0 h, 03c0 h, 06e0 h, 0760 h, 0ee0 h, 
00f0 h, 00f0 h, 01f8 h, 00b0 h, 01f0 h, 00f0 h, 0060 h, 00f0 h, 01d8 h, 01d8 h, 01d8 h, 01b8 h, 00f0 h, 0060 h, 0060 h, 00e0 h, 
003c h, 003c h, 007e h, 002c h, 007c h, 003c h, 0018 h, 003c h, 007e h, 007e h, 00ef h, 00df h, 003c h, 006e h, 0076 h, 00ee h, 
decimal

Crashes most times half way from reboot:

Welcome to zeptoforth
Built for rp2040, version 1.0.2, on Sat Jun 10 05:07:21 PM CDT 2023
zeptoforth comes with ABSOLUTELY NO WARRANTY: for details type `license'
 ok
create animated_jsw  ok
hex  ok
3c00 h, 3c00 h, 7e00 h, 3400 h, 3e00 h, 3c00 h, 1800 h, 3c00 h, 7e00 h, 7e00 h, f700 h, fb00 h, 3c00 h, 7600 h, 6e00 h, 7700 h,  ok
0f00 h, 0f00 h, 1f80 h, 0d00 h, 0f80 h, 0f00 h, 0600 h, 0f00 h, 1b80 h, 1b80 h, 1b80 h, 1d80 h, 0f00 h, 0600 h, 0600 h, 0700 h,  ok
03c0 h, 03c0 h, 07e0 h, 0340 h, 03e0 h, 03c0 h, 0180 h, 03c0 h, 07e0 h, 07e0 h, 0f70 h, 0fb0 h, 03c0 h, 0760 h, 06e0 h, 0770 h,  ok
00f0 h, 00f0 h, 01f8 h, 00d0 h, 00f8 h, 00f0 h, 0060 h, 00f0 h, 01f8 h, 03fc h, 07fe h, 06f6 h, 00f8 h, 01da h, 030e h, 038c h,

tabemann, 2023-10-29:
This seems to be a result of me being too clever in my implementation of USB, such that my code is breaking when it fills up the 128 byte USB Tx buffer; I am trying to simplify my code for such a bit and will see whether it fixes.

tabemann, 2023-10-29:
This turns out to be an off-by-one error - it tries to load up to 128 bytes in the USB Rx buffer, but despite having space allocated for 128 bytes, for reasons (i.e. avoiding maintaining a separate count variable) it only can store 127 bytes in the USB Rx buffer at any time, so it drops the 128th byte in the bit bucket. It just happens that `00f0 h, 00f0 h, 01f8 h, 00d0 h, 00f8 h, 00f0 h, 0060 h, 00f0 h, 01f8 h, 03fc h, 07fe h, 06f6 h, 00f8 h, 01da h, 030e h, 038c h,` is precisely 127 bytes in length, and the CR at the end is the 128th byte, so it gets dropped in the bit bucket, so the line never finishes being parsed.

I have made a fix to this which is now in the `master` and `devel` branches, but is pending being put in a release.

closed 2023-10-29 by tabemann