tabemann/zeptoforth #79: RP2040 USB PLL settings

pkoning2, 2023-12-13:
I was reading the RP2040 initialization code in `src/rp2040/hardware.s` and noticed something: the USB PLL is configured for a VCO frequency of 480 MHz.  The RP2040 data sheet says the VCO frequency is supposed to be between 750 and 1600 MHz.
I have not seen problems from this, but I wonder if it would make sense to change the code to conform to the data sheet requirements.  For example, doubling the PLL frequency would do, that would mean PLL_FBDIV would be 80 rather than 40, and PLL_POSTDIV2 would be 4 rather than 2.

tabemann, 2023-12-13:
Thanks for catching this! I overlooked this when I got the code from its original author. I will make the changes when I come home from work today.

tabemann, 2023-12-14:
I have fixed this issue, and you can get the latest, build 1.3.3.2, from [here](https://github.com/tabemann/zeptoforth/releases/tag/v1.3.3.2).

pkoning2, 2023-12-16:
Looks good, thanks!

closed 2023-12-16 by pkoning2