tabemann/zeptoforth #81: pins-pio-alternate needs a pio argument; sm-sideset-pins! issue related to alternate pins

pkoning2, 2023-12-23:
`pins-pio-alternate` needs a third argument `pio` because the function select code to be used depends on the PIO.  The current code has 6, which is for PIO0, but in my testing I use PIO1 so I get no outputs.  A small change fixes this:
```
  : pins-pio-alternate ( pin-base pin-count pio -- )
    dup validate-pio
    PIO0 = if 6 else 7 then -rot
    dup 30 u<= averts x-too-many-pins
    over 30 u< averts x-pin-out-of-range
    over + swap ?do dup i 30 umod alternate-pin loop
    drop
  ;
```
The documentation also should be corrected; it says that an explicit call is needed for input pins, but that is not true.  Unlike some other I/O devices, PIO always has access to all inputs.  So `pins-pio-alternate` is needed only for outputs (set, out, sideset) where it is already called.
On `sm-sideset-pins!` the way it calls `pins-pio-alternate` with the supplied "pins" count is not right if the sm-sideset-high-enable feature is turned on.  In that case, the value to be written into the "sideset count" register field is the number of pins plus one (for the enable bit) but only the number of pins, not plus one, should be selected for the PIO.  One solution would be to require the caller to do `pins-pio-alternate`, removing it from `sm-sideset-pins!`.  The alternative would be to have a combined operation that is given both the sideset pin count and the high enable flag, so it can select the correct number of pins and also set the correct sideset count.

tabemann, 2023-12-24:
Thanks for catching this! (This is what I get for not spending enough time reading the RP2040 datasheet...)

tabemann, 2023-12-25:
I have addressed these issues in [release 1.3.5](https://github.com/tabemann/zeptoforth/releases/tag/v1.3.5), which modifies `pins-pio-alternate` to take a PIO peripheral, as you recommended, and I also have removed implicit calls to this word from other words, so it will now have to be explicitly specified by the user. (Yes, these are breaking changes for just about anything which uses PIO.)

pkoning2, 2023-12-25:
I suspect there aren't many PIO users, at least not those who beat on it significantly.

BTW, I've been playing with some extensions to the pio module to make a PIO program vaguely object-like.  Among other things, that enables things analogous to the "mark" feature in the ARM assembler module, as well as tagging wrap points the way the Raspberry Pico assembler does it.

closed 2023-12-25 by pkoning2

tabemann, 2023-12-26:
> BTW, I've been playing with some extensions to the pio module to make a PIO program vaguely object-like. Among other things, that enables things analogous to the "mark" feature in the ARM assembler module, as well as tagging wrap points the way the Raspberry Pico assembler does it.

Cool - I await what you come up for it!