tabemann/zeptoforth #82: Bug in pio-instr-relocate-mem!

pkoning2, 2023-12-25:
`pio-instr-relocate-mem` relocates the jumps, but it still writes the program starting at PIO offset zero rather than at the supplied offset, which overwrites some other program.
To fix it, the line that writes an instruction should read:
```      2 pick i + 2 pick INSTR_MEM !```

tabemann, 2023-12-25:
Thanks for catching this - as you can probably guess, I don't bang on PIO that much.

tabemann, 2023-12-25:
I have added your fix to the `devel` branch and will confirm that it works once I get the chance tonight or tomorrow. Then I will merge it into the other branches and make an (RP2040-ONLY) release containing the change.

tabemann, 2023-12-26:
I have made a fix to this, and tested the fix by using the CYW43439 half-duplex SPI PIO program with a different PIO program address other than 0 to blink the LED driven by the CYW43439.

tabemann, 2023-12-26:
This fix is in [release 1.3.5.1](https://github.com/tabemann/zeptoforth/releases/tag/v1.3.5.1).

pkoning2, 2023-12-26:
PIO was the original reason for me to start using the Pico, so I use it a lot.  Thanks for the quick fix.

closed 2023-12-26 by pkoning2