tabemann/zeptoforth #84: RP2040 spi-baud! doesn't look right

pkoning2, 2024-01-08:
I was thinking about the RP2040 rules for the SPI clock, and looked at the logic in spi.fs.  It looks strange and doesn't seem to work right.

The natural thing to do would be to divide the system clock by the desired SPI clock, rounded up, and then use that as the divisor.  For slow SPI clocks that doesn't quite work because there are two dividers (7 and 8 bits respectively).  So I see there is code to calculate the "prescaler" first, then the second divider from that.

But find-spi-prescaler doesn't work right.  If I ask it for the prescaler for 20 MHz it tells me 4, which is an ok answer.  But if I change sysclk to 120 MHz and ask it that same question I should get 2 or 6 but I still get 4 instead, and find-spi-postdiv gives 4 for a SPI clock of 15 MHz rather than 20.  It turns out the reason is that find-spi-prescaler gets a 32 bit overflow so it doesn't find 2 as a valid prescale.

I'm curious why the code works the way it does.

tabemann, 2024-01-08:
I have confirmed that it indeed does have an integer overflow - thanks for
catching this!

Travis

tabemann, 2024-01-09:
I have fixed this issue, and the fix is now in the `devel`, `master`, and `editor-devel` (as I am working on a text editor named `zeptoed`, and it relies on SPI as it edits files in FAT32 filesystems on SDHC cards) branches.