tabemann/zeptoforth #88: Task on core 1 crashes if USB based console output is used

pkoning2, 2024-02-04:
I tried running tasks on core 1 but they failed badly.  Some experiments show that "emit" crashes if the USB console is used.  If I redirect console output to the UART it works fine.

tabemann, 2024-02-05:
I apologize - I didn't initialize the console hooks for the first core 1 task started. Somehow I managed to not notice this until now, probably because in my testing I was always testing with the _second_ core 1 task started for whatever reason. The fix is now in the `main` and `devel` branches.

Travis

tabemann, 2024-02-05:
What is strange about this is that I am very sure I tested this multiple times in the past, and yet I am having trouble finding whether this code had gotten _removed_ at some point in the past...

pkoning2, 2024-02-05:
It's annoying when that happens!

tabemann, 2024-02-06:
I think I figured out what happened ─ I previously had code that initialized peripherals on core 1, and it must have set up the console hooks for USB amongst other things, but when I removed this code as I had found it to cause issues, I forgot to add the console hooks for when core 1 was just started.

BTW, I'm making a new build, build 1.5.0.1, right now that will contain this fix.

tabemann, 2024-02-06:
[Release 1.5.0.1](https://github.com/tabemann/zeptoforth/releases/tag/v1.5.0.1), which fixes this issue, is now out.