tabemann/zeptoforth #94: epub Doc error

CVBruce, 2024-03-03:
In the file zeptoforth-1.5.4/epub/zeptoforth.epub there is the following error message.

“This page contains the following errors:error on line 1506 at column 27: Opening and ending tag mismatch: NL line 1506 and p
Below is a rendering of the page up to the first error.”

Excerpt From
zeptoforth
Travis Bemann
This material may be protected by copyright.

tabemann, 2024-03-03:
Thanks for bringing this to my attention - I don't really look at the epub docs much, and I do not do so on the Mac (I had to borrow my parents' Mac to diagnose this error). It should now be fixed in the `master` and `devel` branches, and the fixed docs will appear in the next release.

closed 2024-03-03 by tabemann

CVBruce, 2024-03-03:
Thanks!Sent from my iPhone

Closed #94 as completed.