tabemann/zeptoforth #97: Sending a buffer over SPI without discarding the result ?

max22-, 2024-04-28:
Hi,
i am trying to port some C code to forth, to read some data from a MCP3008 digital to analog converter.
i need to send 3 bytes, and receive 3. i managed to do it using >spi and spi>, but is there any faster way to do it ?
unfortunately buffer>spi discards the result, and spi>buffer uses a constant filler byte (i need to send 3 different bytes).
thanks
Maxime

tabemann, 2024-04-29:
Are you using the RP2040? Because if you are there is the word buffer>spi-raw-dma ( buffer bytes dma1 dma0 spi -- last-data ). Note that for this word you need to provide your own DMA channels (which can be dynamically allocated with dma-pool::allocate-dma), and it only reports back the last unit of data received.

Travis

max22-, 2024-04-30:
Unfortunately i need the last 2 bytes of data :(

i think in the release this word is not present, if i am right ?
i've cloned the repo and built it from source, to try this word, but when i send setup_full_usb.fs i get this error :
`
\ Iterate executing an xt over a byte array
  : citer ( ??? addr count xt -- ??? ) ( xt: ??? c -- ??? )
    { xt } over + { addr end-addr }
    begin addr end-addr u< while addr c@ xt execute 1 +to addr repeat unable to parse: addr
at Line 70 in File ./src/common/forth/lambda.fs 
`

tabemann, 2024-04-30:
On Tue, Apr 30, 2024 at 4:45 AM Maxime ANDRÉ ***@***.***>
wrote:

> Unfortunately i need the last 2 bytes of data :(
>
Then you are pretty much stuck with >SPI SPI> unless you write a custom
word for this purpose.

> i think in the release this word is not present, if i am right ?
>
Correct, there is currently no built-in word for retrieving the last two
units of data received while sending data; note that one can receive the
last 16 bits of data, but only if one is in 16-bit mode.

> i've cloned the repo and built it from source, to try this word, but when
> i send setup_full_usb.fs i get this error :
> \ Iterate executing an xt over a byte array : citer ( ??? addr count xt --
> ??? ) [...]
How are you doing this exactly, because I have never run into this issue?

Travis

> Message ID: ***@***.***>
>

max22-, 2024-04-30:
ok thank you !

for the error, i have run "make", copied the uf2 on the board (a xiao rp2040), then with e4thcom i've run 
`
#include src/rp2040_big/forth/setup_<type>.fs
`
as mentioned in the readme.
maybe i've done something wrong ?

tabemann, 2024-04-30:
On Tue, Apr 30, 2024 at 2:43 PM Maxime ANDRÉ ***@***.***>
wrote:

> ok thank you !
>
> for the error, i have run "make", copied the uf2 on the board (a xiao
> rp2040), then with e4thcom i've run
> #include src/rp2040_big/forth/setup_<type>.fs
> as mentioned in the readme.
> maybe i've done something wrong ?
>
Which UF2 did you copy onto the board? For this it needs to be an
'rp2040_big' 'kernel' UF2 file.

Also, I recommend you load src/rp2040_big/forth/setup_<type>.fs with:

./utils/codeload3.sh -p <device> -B 115200 serial
src/rp2040_big/forth/setup_<type>.fs

This is what my build scripts use, and I have not had a problem with it
(and it is specially implemented to handle the RP2040 rebooting mid-upload,
which is necessary for building zeptoforth from source).

Travis

max22-, 2024-05-03:
ok now this time it works !
i'll try look at the code to see if i can create a new word that suits to my needs !
thanks !

closed 2024-05-03 by max22-