tabemann/zeptoforth #99: No word being built

twillis-prog, 2024-05-07:
Creating a constant string array. Works in Mecrisp and Gforth.
On Zepoforth zeptoforth-1.5.4.3 - zeptoforth_kernel-1.5.4.3.uf2
Not clear what the issue. is.

  compat import   ok
\ S" ( "ccc<quote>" -- )  ok
: place ( c-addr1 u c-addr2 -- )  ok
  over >r rot over 1+ r> move c! ;  ok
    ok
    ok
: S,  ok
  here over 1+ allot place align ;  ok
  ok
\ parse  ( xchar "ccc<xchar>" – c-addr u  )   ok
: ,"  here 34 parse S, ;  ok
  ok
  ok
: csarray  ( here nstrings -- )  ok
  create  ok
    0 do  ok
            dup , dup c@ 1+ + aligned  ok
    loop  ok
    drop  ok
  does>  ( idx -- addr size )  ok
    swap cells + @ count  ok
;  ok
\ usage  ok
here  ok
," Monday"  ok
," Tuesday"  ok
," Wednesday"  ok
," Thursday"  ok
," Friday"  ok
," Saturday"  ok
," Sunday"  ok
7 csarray day-na2mes no word is being built
Error

tabemann, 2024-05-08:
The issue is that you are using CREATE rather than <BUILDS with DOES> ─ zeptoforth requires <BUILDS with DOES> for reasons (mostly dealing with compilation to flash) whereas gforth uses CREATE ... DOES> and Mecrisp-Stellaris simply has CREATE as a wrapper around <BUILDS which just uses a empty DOES> internally.

Another little note that does not affect this code because you are using the COMPAT module is that one difference between zeptoforth and both gforth and Mecrisp-Stellaris is that ALIGN ( addr|offset power-of-two -- addr'|offset' ) is meant to align an address or offset to a specified power of two, and to get the same behavior in zeptoforth you use CELL ALIGN, ─ however the traditional ANS ALIGN is provided by the COMPAT module which you are using.

Travis

twillis-prog, 2024-05-11:
> Note, GitHub ate my references to <BUILDS with DOES> thinking they were tags, so I am posting this to make sure you notice my comment correctly, which should go like this:

Thank you for clarifying. I have not been able to determine the Dictionary structure with the goal of being able to print the name of a defined constant. The work-around is to build an array of tick addresses. Used Excel to import the rp2040 datasheet table on pages 236/237. Cleaned up imported table and copied into text editor to format to forth.

\  address of word name determined empirically:
: name!     >name 8 + , ;

\ array of GPIO functions
here
create func
' F1  name!
' F2  name!
....

\ array of dictionary pointers
here
create func-name
' CLOCK_GPIN0     name!
' CLOCK_GPIN1     name!
' CLOCK_GPOUT0    name!
 ....

 \ Function Table Cross Reference
here
create func-xref
(    PIN         F1           F2           F3         F4         F5   F6      F7     F8                    F9        )
(  0  $00 )      SPI0_RX   c, UART0_TX  c,  I2C0_SDA c, PWM0_A c,   SIO c, PIO0 c, PIO1 c, NA              c, USB_OVCUR_DET c,
...

\ type function name indexed by i
: .func ( tbl i -- )
   CELLS swap + @ count type
;

: .xref ( func pin -- )
   9 * 1- + func-xref + c@
   func-name swap .func
;

\ Example:
F3 Pin0 .xref I2C0_SDA ok

Competing a set of GPIO tools from 0033mer youtube video, which got me into forth. This was a good exercise.

Thanks for an amazing Forth!

Terry Willis

tabemann, 2024-05-11:
The matter is that there is no way to programmatically print the name of a constant value because any number of `constant`s may share a value. So if you want to print 1 as `F1` you might get some completely unrelated constant `foobar` which just happens to share the value of 1 printed instead.

Travis