tabemann/zeptoforth #101: adc-pin reports "invalid ADC" with default-adc

twillis-prog, 2024-05-14:
Found   ads.fs adc-pin ( adc pin -- ) still had the pin on TOS when validate-adc executed. 
Made to work by adding a  'drop"  prior to validate-adc. When swap was tried, left pin on TOS.

  \ Set a pin to be an ADC pin ( Modified )
  : adc-pin ( adc pin -- )
    dup pin-internal::validate-pin
    dup 26 >= swap 29 <= and averts x-pin-has-no-adc-chan
   drop    \ <<<<<<<<<<<<<< added because the pin was on TOS, causing validate-adc to fail.
    validate-adc
  ;

tabemann, 2024-05-15:
Yeah that is a bug. Actually, if you are targeting the RP2040, don't even use `adc-pin` because it is completely unnecessary. It is there only for compatibility with the STM32 API's sake, and I didn't catch the bug in my testing because I never used `adc-pin` on the RP2040.

twillis-prog, 2024-05-15:
My error! All is OK with adc.fs

closed 2024-05-15 by twillis-prog