tabemann/zeptoforth #123: Zeptofort 1.7.0: Error while writing file with Zeptoed (external memory card)

mcctuxic, 2024-07-28:
Setup:
Linux (GENTOO, Linux kernel 6.10.2)
kitty 0.35.2
zsh 5.9
picocom 3.1
Raspberry Pie Pico W
Zeptoforth 1.7.0

Setting up the Pico W:
Using Zeptoforth 1.7.0 big full usb uf2 to flash the forth system.
entering the forth system via picocom
entering compile-to-flash mode
leaving picocom without resetting the tty (CTRL-a CTRL-q)
from the linux console (kitty/zsh)
utils/codeload3.sh -B 115200 -p /dev/ttyACM0 serial extra/common/fixed32.fs
utils/codeload3.sh -B 115200 -p /dev/ttyACM0 serial extra/common/zeptoed_all.fs
entering the forth system via picocom
compile-to-ram
reboot
entering the forth system via picocom

(after preparing the sd<>-card/the fs)
s" file.fs" zed
<editing text>
CTRL-W
message appears: Wrote file

in the midth of the screen of zed it prints in red
"**heap internal error**"

This happens not all the time but it happens twice this morning after installing Zeptoforth 1.7.0

The file seems to be ok, though.

mcctuxic, 2024-07-28:
I think, I found what trigger the error.

Setup as before.

s" newfile.fs" zed
Added some random lines. Important seems to **add** lines only. **No** removal!
CTRL-W
CTRL-V
no problems so far

s" newfile.fs" zed
now delete some lines
CTRL-W
and when pressing CTRL-V either one gets an "heap internal errror" and then the system hangs hard (need to use the reset button) or it hangs instantly.

mcctuxic, 2024-07-28:
Got

*** HARDWARE EXCEPTION, RETURNING TO PROMPT ***

...what happens here?

mcctuxic, 2024-07-28:
I nuked my board and setup everything again. Cannot say, whether it fixes or changes something yet.
But fscking the memory card, which I freshly partioned and formatted before starting using it with Zeptoforth
gives me this message:

sudo fsck.vfat /dev/sdb1
fsck.fat 4.2 (2021-01-31)
/TEST
  Invalid '..' entry in the second slot. Fixing.
/TEST
  Has a large number of bad entries. (57/64)

/TEST is a directory in the root of that card.

mcctuxic, 2024-07-28:
So...
Summa summarum...

After encountering the above issue and getting the *** HARDWARE EXCEPTION, RETURNING TO PROMPT ***
exception, I nuked the flash of the board and reformatted the memory card, which had fs-error according
to fschk.vfat.

I reinstalled zeptoforth 1.7.0 big full usb uf2.
I installed fixed32 and zeptoed (via the *_all*.fs-file) into flash as described above.
I rewrote block 10...13 to flash from a backup via copy and paste.

The blocks have this content:
B10:

11 13 load-range

B11:

oo import
pin import
simple-fat32 import
fat32-tools import
<simple-fat32-fs> class-size buffer: my-fs
10 11 12 15 1 <simple-fat32-fs> my-fs init-object
my-fs current-fs!
: ls [inlined] list-dir ;
: md [inlined] create-dir ;
: cp [inlined] copy-file ;
: cat [inlined] list-file ;
: rm [inlined] remove-file ;
: clr [inlined] page ;

B12:

: dec [inlined] decimal ;
: oct [inlined] octal ;
: bin [inlined] binary ;
: cls depth 0 ?do drop loop ;

 xon-xoff-enabled
 enable-line
 page

B13:

0 UART::DISABLE-UART
0 output-pin
low 0 pin!

I wrote the files I backuped from the memory card back onto the memory card.

I use 

picocom -f x -b 115200 --logfile zeptolog.txt /dev/ttyACM0

to connect to the board.

But when using zeptoeds CTRL-W to write a changed file I get  "heap internal error" still  when hitting CTRL-V to leave zed.

mcctuxic, 2024-07-28:
Sorry for that many comments Travis, but... :)

I did the same with the previous version (1.6.4.3) and have no problems with zed.
So I think, my hardware is ok.....

tabemann, 2024-07-28:
On Sun, Jul 28, 2024 at 4:06 AM mcctuxic ***@***.***> wrote:

> Sorry for that many comments Travis, but... :)
>
Oh, no need to be sorry.

> I did the same with the previous version (1.6.4.3) and have no problems
> with zed.
> So I think, my hardware is ok.....
>
I was going to say that I was having trouble reproducing the issue... then
I got it to crash. I will be looking into it further.

Travis

> Message ID: ***@***.***>
>

tabemann, 2024-07-28:
On Sun, Jul 28, 2024 at 12:42 PM Travis Bemann ***@***.***> wrote:

> On Sun, Jul 28, 2024 at 4:06 AM mcctuxic ***@***.***> wrote:
>
> Oh, no need to be sorry.
>
> I was going to say that I was having trouble reproducing the issue... then
> I got it to crash. I will be [...]

I was only able to get it to crash once, and since then have had no
problems, and I don't remember what I exactly did when it did crash. Yet
you seem to have been able to get it to crash reliably. Could you write
down your *exact* steps that are triggering a crash, so I can try to
reproduce it reliably?

Thanks,

Travis

> Message ID: ***@***.***>
>>
>

tabemann, 2024-07-28:
On Sun, Jul 28, 2024 at 12:53 PM Travis Bemann ***@***.***> wrote:

> On Sun, Jul 28, 2024 at 12:42 PM Travis Bemann ***@***.***> wrote:
>
>
> I was only able to get it to crash once, and since then have had no
> problems, and I don't remember what I exactly did when it did [...]

Okay, I have made some progress. I have to make files of sufficient size
and then either attempt to close them or close zeptoed. Clearly the heap is
getting corrupted at some point, and the reason why you weren't getting
these errors with 1.6.4.3 is that older versions of zeptoed made no effort
to clean up after themselves as there was no need to close open files.

Travis

> Message ID: ***@***.***>
>>>
>>

tabemann, 2024-07-29:
On Sun, Jul 28, 2024 at 1:34 PM Travis Bemann ***@***.***> wrote:

> On Sun, Jul 28, 2024 at 12:53 PM Travis Bemann ***@***.***> wrote:
>
>
> Okay, I have made some progress. I have to make files of sufficient size
> and then either attempt to close them or close zeptoed. [...]

I am pretty sure I have this sorted now ─ I have made a new release (
https://github.com/tabemann/zeptoforth/releases/tag/v1.7.1) which appears
to fix the zeptoed issues. This release also includes new words for
simplifying accessing files and directories in FAT32 filesystems.

Travis

> Message ID: ***@***.***>
>>>>
>>>

mcctuxic, 2024-07-29:
> I am pretty sure I have this sorted now ─ I have made a new release (
> https://github.com/tabemann/zeptoforth/releases/tag/v1.7.1) which appears
> to fix the zeptoed issues. This release also includes [...]

Hi Travis,

...how many hours of sleep do you get between DO and LOOP or even +LOOP ? :) ;)

An extra big **THANK YOU** for this super fast update, Travis!

I have already updated my two RP2040 boards.

Cheers!
Tuxic

mcctuxic, 2024-07-30:
> 
> > Hi Travis,
> >
> > ...how many hours of sleep do you get between DO and LOOP or even +LOOP ?
> > :) ;)
> >
> 
> More than enough!
> 
> An extra big **THANK YOU** for this super fast update, Travis!
> >
> 
> You're welcome! I try to respond to bug reports in a timely manner.
> 
> I have already updated my two RP2040 boards.
> >
> 
> I may make another release soon even without bug reports because I have
> noticed some weirdness with IMPORT-FROM but I did not want to wait on
> fixing zeptoed.
> 
> Travis

I will preheat my Flash-O-mat, engage the extra fusion power adapter and 
switch on the Zero-point module now to be prepared!

Let's flash!
:)

Cheers!
Tuxic