tabemann/zeptoforth #130: RP2350 support?

extrowerk, 2024-08-08:
https://www.raspberrypi.com/products/raspberry-pi-pico-2/

tabemann, 2024-08-08:
On Thu, Aug 8, 2024 at 10:23 AM extrowerk ***@***.***> wrote:

> https://www.raspberrypi.com/products/raspberry-pi-pico-2/
>
Thank you for bringing this to my attention! I will implement support for
this!

Travis

tabemann, 2024-08-11:
Just for the record, a port of zeptoforth to the RP2350 is in the works as we speak. It currently lives in the `rp2350` branch and covers most of zeptoforth's functionality inherited from the RP2040 so far, but functionality new to the RP2350 such as the Always-On Domain and HSTX is largely missing (but support for QMI is implemented). Note though that I will be unable to test it until my new RP2350 boards, which happen to be on backorder, arrive.

bmentink, 2024-08-23:
This is brilliant! I was just about to ask the same. Will the port use the Hardware Floating point unit and will there be any DSP support?

My boards are on the way, so I will test the branch ASAP.

I take it you are using the ARM cores, would be interesting to know if the RISC cores/instructions are more efficient than ARM, but understand that that would be a huge re-write of the ASM stuff.

znmeb, 2024-08-24:
> This is brilliant! I was just about to ask the same. Will the port use the Hardware Floating point unit and will there be any DSP support?
>
> My boards are on the way, so I will test the branch ASAP. [...]

There are a couple of "Gary Explains" YouTube videos comparing the performance of the RP2040, RP2350 Arm and RP2350 RISC-V cores.. For integer operations it looks like the RP2350 Arm and RISC-V cores are so close in performance that I don't think I'd use a Forth on the RISC-V cores. Also, there's no floating point hardware on the RISC-V cores.

bmentink, 2024-08-24:
> There are a couple of "Gary Explains" YouTube videos comparing the performance of the RP2040, RP2350 Arm and RP2350 RISC-V cores.. For integer operations it looks like the RP2350 Arm and RISC-V [...]

Many Thanks, Understandable.

tabemann, 2024-08-24:
On Fri, Aug 23, 2024, 17:53 bmentink ***@***.***> wrote:

> This is brilliant! I was just about to ask the same. Will the port use the
> Hardware Floating point unit and will there be any DSP support?
>
Hardware floating support may come im the future, but my first priority is
to get what I have already working.

> My boards are on the way, so I will test the branch ASAP.
>
Mind you that it works only as far as that the hardware and flash are
initialized. I am currently having strange problems transmitting data over
UART0 and consequently the serial console does not work yet.

> I take it you are using the ARM cores, would be interesting to know if the
> RISC cores/instructions are more efficient than ARM, but understand that
> that would be a huge re-write of the ASM stuff.
>
What I have heard is that the RISC-V cores are slightly less efficient at
the same clock as the ARM cores, overlooking their lack of hardware
floating point or DSP instructions.

I currently have no plans to support the RISC-V side as a result,
especially since one cannot use both ARM cores and both RISC-V cores at the
same time in the first place.

Travis

>

bmentink, 2024-08-24:
Thanks. However I am having trouble running any of the *.uf2 files even the RP2040 on my RP2040 boards at the moment with the latest git checkout .... is there an issue? I don't get any serial port at all ..

tabemann, 2024-08-24:
On Sat, Aug 24, 2024, 17:42 bmentink ***@***.***> wrote:

> Thanks. However I am having trouble running any of the *.uf2 files even
> the RP2040 on my RP2040 boards at the moment with the latest git checkout
> .... is there an issue? I don't get any serial port at all ..
>
When you say latest git checkout, which branch do you mean? At the moment
the rp2350 branch is purely experimental, even though I could install the
rp2040_big side of it on my RP2040 boards without observing any issues
earlier this week (bit I might have broken it since then).

If you just want to use zeptoforth on an RP2040 build and are not
explicitly intending on testing the rp2350 branch I would recommend using
the latest release tarball or the master branch.

Please let me know what exact configuration you are attempting to use here.

Thanks,
Travis

>

tabemann, 2024-08-25:
On Sat, Aug 24, 2024 at 6:19 PM Travis Bemann ***@***.***> wrote:

> On Sat, Aug 24, 2024, 17:42 bmentink ***@***.***> wrote:
>
> When you say latest git checkout, which branch do you mean? At the moment
> the rp2350 branch is purely experimental, even though I could [...]

Okay, I just tested with the latest code in the rp2350 branch on a generic
Raspberry Pi Pico (not Raspberry Pi Pico 2) hooked up to an SD card adapter
board, and I was not only able to install the latest zeptoforth on it, but
I was able to access the filesystem on my SDHC card without any problems.

Travis

bmentink, 2024-08-25:
I just did a build in "master" and tried the rp2040-big.uf2 on an rp2040 board and it would not present a serial port. WIll try again on my Mac ..

bmentink, 2024-08-25:
I did try the following:
```
./utils/make_uf2_image.sh         
Usage:
    ./utils/make_uf2_image.sh <version> <platform> <port> <image>
No screen session found.
```
But screen is in /opt/homebrew/bin (on MacOS) so it is there, don't know why it was not found ..

tabemann, 2024-08-25:
On Sat, Aug 24, 2024 at 8:14 PM bmentink ***@***.***> wrote:

> I did try the following:
>
> ./utils/make_uf2_image.sh
> Usage:
>     ./utils/make_uf2_image.sh <version> <platform> <port> <image>
> No screen session found.
>
> But screen is in /opt/homebrew/bin (on MacOS) so it is there, don't know
> why it was not found ..
>
> The usage of utils/make_uf2_image.sh is:

make clean
make VERSION=<your version>
make install VERSION=<your version>
<put your board in BOOTSEL mode and flash it with the bin/<your
version>/<your platform>/zeptoforth_kernel-<your version>.uf2 file that was
generated here>
./utils/make_uf2_image.sh <your version> <your platform> <your port> <your
build>

Where <your version> may be something like "1.7.5-dev", where <your
platform> is either "rp2040" or "rp2040_big", <your port> is something like
"/dev/ttyUSB0" (or whatever that would be on macOS), and <your  build> is
one of "mini", "full", or "full_usb". It will place the generated UF2 file
in bin/<your version>/<your platform>/zeptoforth_<your build>-<your
version>.uf2.

Your issue here is that you did not specify any parameters to
./utils/make_uf2_image.sh, hence the message it displayed.

However, this is primarily meant for producing build UF2 files. If you just
want to install zeptoforth on a board without creating a UF2 file, do the
following instead:

make clean
make
<put your board in BOOTSEL mode and flash it with zeptoforth.<your
platform>.uf2>
utils/codeload3.sh -B 115200 -p <your port> serial src/<your
platform>/forth/setup_<your build>.fs

Note that all of the above assumes that your RP2040 board is connected to
your computer via serial (e.g. with a USB-serial dongle); you cannot build
zeptoforth from scratch for the RP2040 using the USB CDC console, as that
is the last thing to be compiled in "full_usb" builds.

Hope this helps,

Travis

> Message ID: ***@***.***>
>

bmentink, 2024-08-25:
Great! Thanks.

bmentink, 2024-09-08:
I tried doing the full build as per your first commands (make_uf2_image.sh) 
I got a lot of forth code being sent to the unit prefixed with TX: but it failed with the line:
` rm: cannot remove 'screenlog:0': No such file or directory`
After a long wait I get:
`00000001ff clone end
No screen session found.`
And it looks like I have a fully built .uf2 file (full_usb in my case) Is this what should happen?

tabemann, 2024-09-08:
On Sun, Sep 8, 2024, 18:03 bmentink ***@***.***> wrote:

> I tried doing the full build as per your first commands (make_uf2_image.sh)
> I got a lot of forth code being sent to the unit prefixed with TX: but it
> failed with the line:
> rm: cannot remove [...]
Oh ignore those messages -- your build succeeded. If it hadn't you wouldn't
have a full_usb .uf2 file.

Travis

>

bmentink, 2024-09-09:
Ok, thanks. I have noticed about an 80% increase in performance in code execution time.
However, my code that works on rp2040 ( A brushless motor control application) only runs for a short while before crashing when executing on rp2350. I presume  it is the silicon pull-down bug, I will investigate ..

tabemann, 2024-09-09:
On Sun, Sep 8, 2024 at 7:51 PM bmentink ***@***.***> wrote:

> Ok, thanks. I have noticed about an 80% increase in performance in code
> execution time.
> However, my code that works on rp2040 ( A brushless motor control
> application) only runs for a short while [...]
This probably isn't the silicon GPIO bug, because that shouldn't cause a
crash. Code that "only runs for a short while before crashing" implies that
garbage is being left on the data stack on the RP2350 but not on the
RP2040. I would be very interested in more details on this issue, so if
there is a bug in the RP2350 port I can fix it.

Travis

tabemann, 2024-09-09:
On Sun, Sep 8, 2024 at 10:15 PM bmentink ***@***.***> wrote:

> Regarding the pull-down bug: It involves a situation where a GPIO pin is
> configured as a pull-down with input buffer enabled.
> If the input is then brought high it can latch to ~2.3v ...
>
> The suggested workaround by Raspberry Pi is to enable the input buffer
> before a read, and disable it again immediately afterwards.
>
> Is this something you might need to add to your IO handling code?
> I am not sure if this effects ADC input ..
>
It turns out that the issue affects not just internal pull-downs with the
input buffer enabled but also floating pins with the input buffer enabled.

I thought of making a workaround but decided against it, because then
people may rely on the workaround's behavior and expect it to behave that
way out of the box, which may cause problems when the silicon bank 0 pad
bug is fixed in a future respin.

Apparently this does not affect ADC input, but does affect anything that
relies on the bank 0 pads being used in an internal pull-down or floating
fashion, including PIO.

Travis

bmentink, 2024-09-09:
Hi Travis,

I have noticed latch up on the ADC's with high impedance source. Other's have as well see: https://forums.raspberrypi.com/viewtopic.php?t=375631

tabemann, 2024-09-09:
On Mon, Sep 9, 2024 at 4:48 PM bmentink ***@***.***> wrote:

> Hi Travis,
>
> I have noticed latch up on the ADC's with high impedance source. Other's
> have as well see: https://forums.raspberrypi.com/viewtopic.php?t=375631
>
I stand corrected - ugh. While I have my RP2350's for the purpose of
getting zeptoforth ported, there are too many issues introduced by the
silicon GPIO pad bug to really recommend it unless one is careful to limit
oneself to applications where one does not need floating pins (or cannot
tolerate the expanded BOM for external pull-down resistors) or, as you
mention, ADC.

Travis

bmentink, 2024-09-09:
No problem. My application works fine, just have to have a low impedance on my throttle pot, and all is well. Everything else runs the same as rp2040, except faster ..

tabemann, 2024-09-09:
On Mon, Sep 9, 2024 at 5:26 PM bmentink ***@***.***> wrote:

> No problem. My application works fine, just have to have a low impedance
> on my throttle pot, and all is well. Everything else runs the same as
> rp2040, except faster ..
>
I thought you said you had problems with crashing on the RP2350. I am
interested in that, to see if any issues got introduced in the RP2350 port
that need fixing.

Travis

bmentink, 2024-09-09:
Sorry, I mistook the adc lockup with a crash, all good so far..