tabemann/zeptoforth #132: codeload3 fails part-way through basic.fs

dendrown, 2024-08-11:
Hello.  Apologies if this is a N00b question.  I am just getting started with zeptoforth.  My problem seems similar to issue #53, but I am not using e4thcom.

The problem occurs with both `rp2040` and `rp2040_big`.  Unlike the e4thcom issue, it fails every time. 

After flashing `bin/1.7.4/rp2040/zeptoforth_full_usb-1.7.4.uf2` from the release tarball, I run this command:
```bash
sudo utils/codeload3.sh -p /dev/ttyACM0 serial src/rp2040/forth/setup_full_usb.fs
```
It transmits a good amount of code to the pi pico, but always fails in the same place:
```
TX: 
TX: \ Is halfword in array
TX: : half-in-array ( value addr count -- in-array? )
TX: 2* over + swap ?do dup i h@ = if drop true unloop exit then 2 +loop
error 2* over + swap ?do dup i h@ = if drop true unloop exit then 2 +loop loop variable not found

Error file src/rp2040/forth/setup_full_usb.fs line 26: could not upload file
>>>  #include src/common/forth/basic.fs
```

If I do the same with `rp2040_big`, I get the same error in the exact same place.

Even with the failure, I am able to connect to the pico via minicom and run some basic Forth commands in the REPL, including the LED demo, but looking through the source code, I see that what successfully loaded barely scratches the surface.

Thanks in advance for your help.

--den

tabemann, 2024-08-11:
On Sun, Aug 11, 2024 at 4:36 PM den Drown ***@***.***> wrot

> Hello. Apologies if this is a N00b question. I am just getting started
> with zeptoforth. My problem seems similar to issue #53
> <https://github.com/tabemann/zeptoforth/issues/53>, but I am not using
> e4thcom.
>
> The problem occurs with both rp2040 and rp2040_big. Unlike the e4thcom
> issue, it fails every time.
>
> After flashing bin/1.7.4/rp2040/zeptoforth_full_usb-1.7.4.uf2 from the
> release tarball, I run this command:
>
> sudo utils/codeload3.sh -p /dev/ttyACM0 serial src/rp2040/forth/setup_full_usb.fs
>
> It transmits a good amount of code to the pi pico, but always fails in the
> same place:
>
> TX:
> TX: \ Is halfword in array
> TX: : half-in-array ( value addr count -- in-array? )
> TX: 2* over + swap ?do dup i h@ = if drop true unloop exit then 2 +loop
> error 2* over + swap ?do dup i h@ = if drop true unloop exit then 2 +loop loop variable not found
>
> Error file src/rp2040/forth/setup_full_usb.fs line 26: could not upload file
> >>>  #include src/common/forth/basic.fs
>
> If I do the same with rp2040_big, I get the same error in the exact same
> place.
>
> Even with the failure, I am able to connect to the pico via minicom and
> run some basic Forth commands in the REPL, including the LED demo, but
> looking through the source code, I see that what successfully loaded barely
> scratches the surface.
>
> Thanks in advance for your help.
>
> --den
>
The reason for your issue is that you are loading
src/rp2040/forth/setup_full_usb.fs on top of an existing build of the same.
src/rp2040/forth/setup_full_usb.fs is meant to be loaded onto a bare
zeptoforth kernel, found in bin/1.7.4/rp2040/zeptoforth_kernel-1.7.4.uf2
(or replace "1.7.4" with the relevant version and "rp2040" with
"rp2040_big" if you are using an rp2040_big build), or a build where the
compiled zeptoforth environment on top of the kernel has been erased with
ERASE-ALL . src/common/forth/basic.fs expects to start out with a bare
kernel and nothing more, and will break if that is not true.

Note also that when loading src/rp2040/forth/setup_full_usb.fs (or like)
you need to use a serial line hooked up to UART0 on GPIO's 0 and 1. It will
not work with USB because it reboots the system mid-load, which makes the
/dev/ttyACM0 (or like) go away momentarily.

Travis

> Message ID: ***@***.***>
>

tabemann, 2024-08-11:
On Sun, Aug 11, 2024 at 5:00 PM den Drown ***@***.***> wrote:

> Ack, I think the full_* bin/uf2 files already have the Forth code
> compiled in, and so I don't need to run the codeload3 utility. Sorry, I
> fear I have made things much more complicated than necessary, getting
> started here.
>
> That is true ─ the bin/1.7.4/rp2040/zeptoforth_full_usb-1.7.4.uf2 (and
like) files already contain a full zeptoforth environment, so there is no
need to use utils/codeload3.sh. utils/codeload3.sh is needed if you are
creating a new build from scratch, e.g. you have modified source files that
are included in zeptoforth builds themselves.

Travis

> Message ID: ***@***.***>
>

dendrown, 2024-08-11:
Thank you for the quick answer on this.  It dawned on me that the LED demo code came way after **basic.fs**, and so what I was running must already be on the Pico.  That plus some targeted re-reading of the main README got me back on track.

If nothing else, I got a preliminary lesson in what I'll need to do when I've got some of my own Forth code I want to load onto the device.

Thanks again for you help and also for your work on the zeptoforth project itself.  It's exciting to have such a full-featured Forth available for the Pi and other microcontrollers.

closed 2024-08-11 by dendrown

tabemann, 2024-08-11:
You're welcome!

Travis