tabemann/zeptoforth #135: Forth beginner whoes

extrowerk, 2024-08-26:
Hi,

I am trying to implement this algorythm to calculate the PI:

[image]

This is the solution I came up with:

```
VARIABLE A
VARIABLE B
VARIABLE C
VARIABLE N
VARIABLE O
VARIABLE P
0 P !
0 O !
1 N !
1 A !
1 B !
: _30A O @ P ! ;
: _30B N @ DUP * N ! ;
: _40 P @ N @ A @ B @ * * + P ! ;
: _50A A @ DUP * B @ DUP * + 4 / ;
: _50B C @ SQRT A ! ;
: _60 1 @ C - SQRT 1 SWAP - B ! ;
: ?BIGGER_THAN_O O @ > IF _30A _30B THEN
: PI  10 0 DO _30A _30B _40 _50A _50B _60 P @ ?BIGGER_THAN_O P @ CR . LOOP ;
```

However it doesn't works at all, I simply get 
```
 unable to parse: ?BIGGER_THAN_O
```

Can you please point out, what am I doing wrong?

tabemann, 2024-08-27:
On Mon, Aug 26, 2024 at 4:08 PM extrowerk ***@***.***> wrote:

> Hi,
>
> I am trying to implement this algorythm to calculate the PI:
>
> image.png (view on web)
> <https://github.com/user-attachments/assets/fd5bb464-222b-4550-ab30-6c97840abd0e>
>
> This is the solution I came up with:
>
> VARIABLE A
> VARIABLE B
> VARIABLE C
> VARIABLE N
> VARIABLE O
> VARIABLE P
> 0 P !
> 0 O !
> 1 N !
> 1 A !
> 1 B !
> : _30A O @ P ! ;
> : _30B N @ DUP * N ! ;
> : _40 P @ N @ A @ B @ * * + P ! ;
> : _50A A @ DUP * B @ DUP * + 4 / ;
> : _50B C @ SQRT A ! ;
> : _60 1 @ C - SQRT 1 SWAP - B ! ;*: ?BIGGER_THAN_O O @ > IF _30A _30B THEN*
> : PI  10 0 DO _30A _30B _40 _50A _50B _60 P @ ?BIGGER_THAN_O P @ CR . LOOP ;
>
> However it doesn't works at all, I simply get
>
>  unable to parse: ?BIGGER_THAN_O
>
> Can you please point out, what am I doing wrong?
>
> The issue was that you forgot the ';' at the end of the line on which you
defined ?BIGGER_THAN_O so it never got finished being defined (and thus
could not be looked up). Note that it *is* legal to use ':' within a word
definition, which is why you did not get an immediate error (and it just
happens that PI is an already existing word, which is why no error occurred
after that either).

Travis

> Message ID: ***@***.***>
>

extrowerk, 2024-08-27:
Thanks for the feedback, I have made the following modifications:
- added semicolon to ?BIGGER_THAN_O definition
- renamed PI to PII
Now calling PII results a hardware exception.

```
Welcome to zeptoforth
Built for rp2040_big, version 1.7.4, on Wed Aug 7 08:11:39 PM CDT 2024
zeptoforth comes with ABSOLUTELY NO WARRANTY: for details type `license'
 ok
VARIABLE A  ok
VARIABLE B  ok
VARIABLE C  ok
VARIABLE N  ok
VARIABLE O  ok
VARIABLE P  ok
0 P !  ok
0 O !  ok
1 N !  ok
1 A !  ok
1 B !  ok
: _30A O @ P ! ;  ok
: _30B N @ DUP * N ! ;  ok
: _40 P @ N @ A @ B @ * * + P ! ;  ok
: _50A A @ DUP * B @ DUP * + 4 / ;  ok
: _50B C @ SQRT A ! ;  ok
: _60 1 @ C - SQRT 1 SWAP - B ! ;  ok
: ?BIGGER_THAN_O O @ > IF _30A _30B THEN ;  ok
: PII 10 0 DO _30A _30B _40 _50A _50B _60 P @ ?BIGGER_THAN_O P @ CR . LOOP ;  ok
PII
*** HARDWARE EXCEPTION, RETURNING TO PROMPT ***
```

extrowerk, 2024-08-27:
I had a typo in line 60, fixed like this:

```
: _60 1 C - SQRT 1 SWAP - B ! ; 
```

Now it doesn't crashing anymore, but the printed results doesn1t makes any sense. I suspect the 0-10 loop is just silly, the original algorythm nowhere asks for this, so I have to think about this further. Thanks for your help so far!

tabemann, 2024-08-27:
On Tue, Aug 27, 2024 at 4:52 AM extrowerk ***@***.***> wrote:

> I had a typo in line 60, fixed like this:
>
> : _60 1 C - SQRT 1 SWAP - B ! ;
>
> Now it doesn't crashing anymore, but the printed results doesn1t makes any
> sense. I suspect the 0-10 loop is just silly, [...]
One thing to remember is that SQRT acts on S31.32 fixed-point values, which
are double cells. Therefore you need to have all your logic designed to use
double cells, including using 2VARIABLE's, 2@, 2!, 2SWAP, D+, D-, F*, F/,
fixed point constants (e.g. 1,0 rather than 1), and like. Also you will
want your output to be an S31.32 fixed-point value, because it does not
make much sense to calculate the value of pi as an integer. (Note that to
print a S31.32 fixed-piont value you use F.)

Also note that in the above you forgot a @ even if intended to use normal
single-cell integers; the @ would go after the C.

I would rewrite _60 as:

: _60 1,0 C 2@ D- SQRT 1,0 2SWAP D- B 2! ;

Travis

extrowerk, 2024-08-30:
Made it this far:

```
2VARIABLE A
2VARIABLE B
2VARIABLE C
2VARIABLE N
2VARIABLE O
2VARIABLE P
0,0 P 2!
0,0 O 2!
1,0 N 2!
1,0 A 2!
1,0 B 2!

( O = P )
: _30A O 2@ P 2! ;

( N=N*2 )
: _30B N 2@ 2,0 F* N 2! ;

( P=P+N*A*B )
: _40 N 2@ A 2@ B 2@ F* F* P 2@ D+ P 2! ;

( C=(A*A+B*B)/4 )
: _50A A 2@ 2DUP F* B 2@ 2DUP F* D+ 4,0 F/ C 2! ;

( A=SQR(C)  )
: _50B C 2@ SQRT A 2! ;

( B=1-SQR(1-C)  )
: _60 1,0 C 2@ D- SQRT 1,0 2SWAP D- B 2! ;

( goto 30 )
: GOTO30 _30A _30B _40 _50A _50B _60 ;

( IF P>O then )
: ?BIGGER_THAN_O O 2@ > IF GOTO30 THEN ;

( =  )
: PII 10 0 DO P 2@ ?BIGGER_THAN_O P 2@ CR F. LOOP ;
```

Still not working but i start to see tje light..

tabemann, 2024-08-30:
On Thu, Aug 29, 2024 at 10:34 PM extrowerk ***@***.***> wrote:

> Made it this far:
>
> 2VARIABLE A
> 2VARIABLE B
> 2VARIABLE C
> 2VARIABLE N
> 2VARIABLE O
> 2VARIABLE P
> 0,0 P 2!
> 0,0 O 2!
> 1,0 N 2!
> 1,0 A 2!
> 1,0 B 2!
>
> ( O = P )
> : _30A O 2@ P 2! ;
>
> ( N=N*2 )
> : _30B N 2@ 2,0 F* N [...]
The following line needs to be changed:

: ?BIGGER_THAN_O O 2@ *>* IF GOTO30 THEN ;

This should be:

: ?BIGGER_THAN_O O 2@ *D>* IF GOTO30 THEN ;

> uses single-cell values, D> uses double-cell values and supports S31.32
fixed-point numbers.

Travis

> Message ID: ***@***.***>
>

extrowerk, 2024-09-06:
This is not an issue with ZF, but with my knowledge about Forth, so lets close it for now. Sorry for the noise.

closed 2024-09-06 by extrowerk

tabemann, 2024-09-06:
On Fri, Sep 6, 2024 at 3:34 AM extrowerk ***@***.***> wrote:

> This is not an issue with ZF, but with my knowledge about Forth, so lets
> close it for now. Sorry for the noise.
>
No, it's perfectly fine. I like helping out people with their Forth issues,
even when they are not issues with zeptoforth. And in this case, while
double-cell arithmetic is not unique to zeptoforth, some aspects of this,
particularly the fixed-point arithmetic, are more specific to zeptoforth
and Mecrisp-Stellaris.

Travis