tabemann/zeptoforth #149: Unreliable USB CDC on RP2040?

extrowerk, 2024-10-17:
Hi,

I have various RP2040 based boards from Aliexpress & Tindie and I experience issues with all of them which is suspicious.

I have tested the boards with:
- different USB cables
- on different computers
- using different OS (OpenBSD & Windows)
- with different baud-speed setting defined in Zeptoforth (even as low as 9600 baud)
- using different serial terminals (cu, minicom, putty)
- with the latest Zeptoforth release

Still, a simple "counting upwards and printing all the numbers (i) and its squared counterpart (i*i) frequently results in deadlock.
From memory, the algorithm is something like this:

```
_ii_ : dup * . ;
_separator_ : 58 emit ;
_print_ : 0 do i dup . _separator_ _ii_ cr ;
```

It should print sequences like the following:
1 : 1
2 : 4
3 : 9
4 : 16
and so on.

In my case, this goes on for a while, but frequently it freezes after a some iteration.

I haven't tested the UART pins on the boards.

I have just received one board ( https://de.aliexpress.com/item/1005004939480808.html ), which shows up as a normal Pico USB drive, but copying the "full usb" uf2 file it doesn't even shows up as a serial port.

Are my boards busted? Counterfeit maybe? Or am I doing something wrong?

znmeb, 2024-10-17:
Do the boards work in MicroPython and CircuitPython? That would be the first thing I'd try. If the hardware is flaky my guess is at least one of the two embedded Python tools will croak. If they both work, try "Hello USB" in the SDK examples.

extrowerk, 2024-10-17:
The tiny board became alive as soon as I have copied the micropython uf2, but I have to do dive into micropython and circuitpython for properly test the boards.
Nevertheless after that I have tried again with Zeptoforth and now it shows up as serial port and I can access the Forth REPL.

I'll report back with the testresults.

extrowerk, 2024-10-17:
I have to add: I have experienced the CDC serial issues with the raw board no external devices connected to it, so it is not a power supply issue in my POV.

znmeb, 2024-10-17:
> The tiny board became alive as soon as I have copied the micropython uf2, but I have to do dive into micropython and circuitpython for properly test the boards. Nevertheless after that I have tried [...]

See also "flash_nuke.uf2" https://forum.micropython.org/viewtopic.php?t=10838#p59604

tabemann, 2024-10-17:
On Thu, Oct 17, 2024 at 9:41 AM extrowerk ***@***.***> wrote:

> : *ii* dup * . ;
> : *separator* 58 emit ;
> : *print* 0 do i dup . *separator* *ii* cr loop ;
>
> Yes, this works for me aswell. However as I explained, i wanted to see how
> fast the it finishes without printing on the serial port, so I have removed
> al lthe printing operations, and reduced the program to:
>
>  : _ii_ dup * ;
>  : this-will-not-print 0 do i dup _ii_ loop ;
>  150000 this-will-not-print
>
> And it freezes. Probably I am doing something wrong, but my naive mind
> says, it should just calculate the value pairs and...
>
> Wait a sec!!!! Now I see, it fills up the stack and then it freezes
> probably! Then it is my fault....
>
Exactly -- it will explode the data stack, overwriting the task structure,
crashing the system unrecoverably. Your code should read:

 : _ii_ dup * drop ;
 : this-will-not-print 0 do i dup _ii_ loop ;
 150000 this-will-not-print

Did you still have problems with locking up mid-transfer with USB
though? If you did, I would be very much interested in such.

Travis

extrowerk, 2024-10-18:
I will let you know if I see nay lockup other than this.

closed 2024-10-18 by extrowerk