tabemann/zeptoforth #160: s31.32 fixed-point numbers calculation ?

meineid-git, 2024-12-03:
I just poked around with zeptoforth built V1.9.0 for rpi2350 and it appears to me as if the s31.32 feature is broken:

`2,0 f. 2,1250086035  ok
2,0 3,0 d+ f. 5,2500172080  ok
3,1415927 f. 3,1250086035  ok`

tabemann, 2024-12-03:
On Tue, Dec 3, 2024 at 4:34 AM meineid-git ***@***.***> wrote:

> I just poked around with zeptoforth built V1.9.0 for rpi2350 and it
> appears to me as if the s31.32 feature is broken:
>
> 2,0 f. 2,1250086035 ok 2,0 3,0 d+ f. 5,2500172080 ok 3,1415927 f.
> 3,1250086035 ok
>
> Thanks for finding that - on the RP2350 I get:

2,0 f. 2,1250086035  ok

but on the RP2040 I get:

 2,0 f. 2,0  ok

Also, I have confirmed that it is the literal parsing that is at fault
rather than the numeric formatting.

Travis

> Message ID: ***@***.***>
>

tabemann, 2024-12-03:
On Tue, Dec 3, 2024 at 9:30 AM Travis Bemann ***@***.***> wrote:

> On Tue, Dec 3, 2024 at 4:34 AM meineid-git ***@***.***>
> wrote:
>
>
> 2,0 f. 2,1250086035  ok
>
> but on the RP2040 I get:
>
>  2,0 f. 2,0  ok
>
> Also, I have confirmed that it is the literal parsing that is at [...]

I also just confirmed that it only is broken when in interpretation mode
(probably how I missed it), as shown by:

Welcome to zeptoforth
Built for rp2350, version 1.9.0, on Thu Nov 14 01:33:19 PM CST 2024
zeptoforth comes with ABSOLUTELY NO WARRANTY: for details type `license'
 ok
: foo 2,0 f. ;  ok
foo 2,0  ok

Travis

> Message ID: ***@***.***>
>>
>