tabemann/zeptoforth #168: How do we suppose to handle user input?

extrowerk, 2024-12-31:
I am trying to learn Zeptoforth doing excercises, one common thing is handling user-inputs, like the following:

https://github.com/migiFi/C-exercises/tree/main/exercises/exercise_2_2

I have looked at the Forth documentations, but it seems the ACCPET word doesn't exists in Zeptoforth, and I haven't found any hints in the Zeptoforth docs folder either.

Can somebody point me in the correct direction?

While i could use the stack to pass the input value here first, then call my word, but I'd like to have exactly the same output like it is described in the excercise.

Thanks for your help!

tabemann, 2024-12-31:
On Tue, Dec 31, 2024 at 4:33 AM extrowerk ***@***.***> wrote:

> I am trying to learn Zeptoforth doing excercises, one common thing is
> handling user-inputs, like the following:
>
> https://github.com/migiFi/C-exercises/tree/main/exercises/exercise_2_2
>
> I have looked [...]
Yes, zeptoforth doesn't have ACCEPT at the present. However, that doesn't
stop one from writing it. I decided to write and test a simple ACCEPT,
which is as follows:

$08 constant c-backspace
$0A constant c-linefeed
$0D constant c-carriage-return
$7F constant c-delete

: accept { addr bytes -- bytes' }
  addr { current-addr }
  begin
    key
    dup c-linefeed = over c-carriage-return = or if
      drop current-addr addr - cr exit
    else
      dup c-backspace = over c-delete = or if
        drop
        current-addr addr u> if
          -1 +to current-addr
          c-backspace emit space c-backspace emit
        then
      else
        current-addr addr bytes + u< if
          dup emit
          current-addr c!
          1 +to current-addr
        else
          drop
        then
      then
    then
  again
;

Note that this ACCEPT does not provide any fancy features -- no proper line
editing, just entering characters and deleting them in reverse order
entered. It does not play nice with editing past the end of the current
line in one's terminal, as is typical of most naive text line entry.

Part of the reason I never bothered to implement ACCEPT is that for most
plain entry I just use the REPL, and when I need more than the REPL I
usually need something specialized, e.g. in the case of zeptoed. However,
as obviously there is demand for ACCEPT I will include this in future
releases of zeptoforth. It will probably not be this version, however,
because I will make it properly support line editing when ENABLE-LINE has
been specified.

Travis

> Message ID: ***@***.***>
>

tabemann, 2024-12-31:
On Tue, Dec 31, 2024 at 1:29 PM Travis Bemann ***@***.***> wrote:

> On Tue, Dec 31, 2024 at 4:33 AM extrowerk ***@***.***>
> wrote:
>
> Yes, zeptoforth doesn't have ACCEPT at the present. However, that doesn't
> stop one from writing it. I decided to write and test a [...]

I will soon be including this functionality, including the ability to use
the line editor when ENABLE-LINE has been specified, in the full and devel
branches. Note, however, there will be one minor difference ─ I will not be
automatically outputting a CRLF when the user presses enter, to make the
functionality more consistent with the zeptoforth REPL, where pressing
enter does not, by default, output CRLF.

Travis

> Message ID: ***@***.***>
>>
>

tabemann, 2024-12-31:
On Tue, Dec 31, 2024 at 2:38 PM Travis Bemann ***@***.***> wrote:

> On Tue, Dec 31, 2024 at 1:29 PM Travis Bemann ***@***.***> wrote:
>
>
> I will soon be including this functionality, including the ability to use
> the line editor when ENABLE-LINE has been specified, in [...]

I should note that the code above does not ignore escape, that is, $1B, so
it behaves funny if one enters arrow keys or like while using it. In the
latest code in master and devel this issue has been fixed (where it outputs
garbage when entering arrow keys if line editing is not enabled, but at
least it does not have very non-obvious behavior).

Travis

> Message ID: ***@***.***>
>>>
>>