tabemann/zeptoforth #181: Rp2350B "HI' pins not working

amken3d, 2025-02-12:
After spending a day trying to figure out why SPI was not working, I discovered that the HI (pins 32-48) are not functioning. I validated that it is not a problem with my board by running code to toggle pins using the Pico SDK. A cursory glance at pins.fs did not indicate anything wrong. However, there may be some conflict somewhere. I am not familiar enough  with Zeptoforth to be able to figure it out yet. I am using the full usb uf2 from the latest release.

tabemann, 2025-02-12:
On Wed, Feb 12, 2025 at 1:45 PM Amken USA ***@***.***> wrote:

> After spending a day trying to figure out why SPI was not working, I
> discovered that the HI (pins 32-48) are not functioning. I validated that
> it is not a problem with my board by running code to [...]

Could you paste a code example? I would like to see what is going on with
your code (particularly since while I have a RP2350B board I have no
convenient way to get at the HI GPIO's other than to use PSRAM tied to GPIO
47).

Travis

tabemann, 2025-02-13:
On Wed, Feb 12, 2025 at 5:33 PM Travis Bemann ***@***.***> wrote:

> On Wed, Feb 12, 2025 at 1:45 PM Amken USA ***@***.***>
> wrote:
>
>
> Could you paste a code example? I would like to see what is going on with
> your code (particularly since while I have a RP2350B board I [...]

I am kind of baffled by this, because the only place there could be a
problem is in PIN::ALTERNATE-PIN (which is called by SPI::SPI-PIN) and the
GPIO words it calls, but I carefully looked those over and they all look
perfectly fine to me. I even checked the RP2350 datasheet to make sure the
same function is used for SPI for GPIO's 32-47 as with GPIO's 0-31, and
there is no difference between the two sets of GPIO's.

Travis

amken3d, 2025-02-13:
This was what I was trying to run 

marker reset-ui  
pin import  
spi import  
ansi-term import  
: color-red ( -- ) escape csi ." 31m" 
: color-green ( -- ) escape csi ." 32m" 
: color-yellow ( -- ) escape csi ." 33m" 
: color-blue ( -- ) escape csi ." 34m" 
: color-cyan ( -- ) escape csi ." 36m" 
: color-reset ( -- ) escape csi ." 0m" 
1 constant spi1  
42 constant motor_sck  
43 constant motor_mosi  
40 constant motor_miso  
47 constant motor_cs1  
8  constant motor_cs2  
11 constant motor_cs3  
21 constant motor_cs4  
20 constant motor_cs5  
24 constant motor_cs6  
27 constant motor_cs7  
28 constant motor_cs8  
: init-spi1 ( -- )  
page  
hide-cursor  
cr ." Initializing SPI1..." cr  
spi1 motor_sck spi-pin  
spi1 motor_mosi spi-pin  
spi1 motor_miso spi-pin  
motor_cs1 output-pin high motor_cs1 pin!  
motor_cs2 output-pin high motor_cs2 pin!  
motor_cs3 output-pin high motor_cs3 pin!  
motor_cs4 output-pin high motor_cs4 pin!  
motor_cs5 output-pin high motor_cs5 pin!  
motor_cs6 output-pin high motor_cs6 pin!  
motor_cs7 output-pin high motor_cs7 pin!  
motor_cs8 output-pin high motor_cs8 pin!  
spi1 disable-spi  
spi1 master-spi  
1000000 spi1 spi-baud!  
spi1 3 0 motorola-spi  
8 spi1 spi-data-size!  
spi1 enable-spi  
show-cursor  
cr ." SPI1 initialized successfully." cr  

: select-motor ( cs-pin -- )  
low swap pin!  

: deselect-motor ( cs-pin -- )  
high swap pin!  

: spi-write ( byte -- ) spi1 >spi 
: spi-read ( -- byte ) spi1 spi> 
: spi-transfer40 ( reg cs-pin -- response )  
dup select-motor  
spi-write  
0 spi-write  
dup deselect-motor  
1. timer::delay-us  
dup select-motor  
spi-write  
spi-read 24 lshift  
spi-read 16 lshift or  
spi-read 8 lshift or  
spi-read or  
dup deselect-motor  

: tmc5160-read ( reg cs-pin -- value )  
>r  \ Store CS pin on return stack  
cr color-blue ." [DEBUG] Selecting CS: " r@ . cr  
r@ low  \ Select chip (CS low)  
cr color-yellow ." [DEBUG] Sending read command for $" over h.2 cr  
7 lshift 0 or spi1 >spi  \ Send register read command  
cr color-cyan ." [DEBUG] Receiving data..." cr  
spi1 spi> 24 lshift  
spi1 spi> 16 lshift or  
spi1 spi> 8 lshift or  
spi1 spi> or  
cr color-blue ." [DEBUG] Deselecting CS" cr  
r> high  \ Deselect chip (CS high)  

: dumpTMC ( motor -- )  
page hide-cursor  
color-blue  
cr ." +--------------------------------+" cr  
." |  Register  |   Value (HEX)     |" cr  
." +--------------------------------+" cr  
color-reset  
dup case  
1 of motor_cs1 endof  
2 of motor_cs2 endof  
3 of motor_cs3 endof  
4 of motor_cs4 endof  
5 of motor_cs5 endof  
6 of motor_cs6 endof  
7 of motor_cs7 endof  
8 of motor_cs8 endof  
( Default ) drop 0  
endcase  
dup 0= if  
color-red ." Invalid motor selection!" cr color-reset exit  
then  
12 0 do  
color-yellow ." |  $" i h.2 space space  \ Register number  
dup i tmc5160-read  \ Read register  
color-cyan ." |   $" h.8 space space ."  |" cr  \ Display value  
loop  
color-blue  
cr ." +--------------------------------+" cr  
color-green  
." Test complete." cr  
color-reset show-cursor  

: display-motor-selection ( -- )  
page hide-cursor  
color-red  
cr ." +----------------------------+" cr  
." |   Select a Motor (1-8)     |" cr  
." +----------------------------+" cr  
color-green  
cr ." |  1 - Motor 1 | 5 - Motor 5 |" cr  
." |  2 - Motor 2 | 6 - Motor 6 |" cr  
." |  3 - Motor 3 | 7 - Motor 7 |" cr  
." |  4 - Motor 4 | 8 - Motor 8 |" cr  
color-red  
cr ." +----------------------------+" cr  
color-reset  
cr ." Enter motor number: " show-cursor  

$00 constant gconf  
$01 constant gstat  

I then used something like this to test each pin 

pin import
: b0 ( -- )
     0 output-pin  \ Set pin as output

    begin
            low 0 pin!
                100 ms  \ Delay between toggles
                high 0 pin!
                100 ms  \ Delay between toggles
    again
;

amken3d, 2025-02-13:
I was baffled as well as there wasnt anything obvious in pin.fs

tabemann, 2025-02-13:
Is there a reason there is not a ; after each word definition? If there
isn't one, that would definitely cause problems. (zeptoforth is not
colorForth -- it requires ; after word definitions before the next
word-definition.)

Travis

tabemann, 2025-02-13:
I should note, however, that : is not an immediate word in zeptoforth,
which allows tricks like:

: make-adder ( n "name" -- )
  : immediate postpone literal postpone + postpone ;
;

which you use like:

4 make-adder 4+

which creates a word 4+ ( n -- n' ) which adds 4 to the top of the stack.

However, while : is not immediate in ANS Forth, this has odd results when
you try to use it this way if you do not expect it.

Travis

amken3d, 2025-02-14:
Sorry, I am a total noob. Until this week I did not even know the existence of Forth. I just happened to stumble into Forth when I was trying to figure out a way to build a REPL for Tinygo. 
 I was trying to experiment, by following examples and the api  to come up with the code.  I am amazed at the what it can do, I was able to, with a bit of guessing, figure out how to use Zeptoforth.  But easy to use examples, and some hand holding would go a long way in bringing in new users. FWIW, I think it provides a great alternative to using C/SDK.  The REPL is just mwah... 
As for what you replied. I will be honest, I did not understand a single thing. But that is not on you. :)

tabemann, 2025-02-14:
> Sorry, I am a total noob. Until this week I did not even know the
> existence of Forth. I just happened to stumble into Forth when I was trying
> to figure out a way to build a REPL for Tinygo.
> I was trying to experiment, by following examples and the api to come up
> with the code. I am amazed at the what it can do, I was able to, with a bit
> of guessing, figure out how to use Zeptoforth. But easy to use examples,
> and some hand holding would go a long way in bringing in new users. FWIW, I
> think it provides a great alternative to using C/SDK. The REPL is just
> mwah...
> As for what you replied. I will be honest, I did not understand a single
> thing. But that is not on you. :)
>
>
> Sorry, I am a total noob. Until this week I did not even know the
> existence of Forth. I just happened to stumble into Forth when I was trying
> to figure out a way to build a REPL for Tinygo.
> I was trying to experiment, by following examples and the api to come up
> with the code. I am amazed at the what it can do, I was able to, with a bit
> of guessing, figure out how to use Zeptoforth. But easy to use examples,
> and some hand holding would go a long way in bringing in new users. FWIW, I
> think it provides a great alternative to using C/SDK. The REPL is just
> mwah...
> As for what you replied. I will be honest, I did not understand a single
> thing. But that is not on you. :)
For your code, after defining each word ("word" in this context is
Forth-speak for what other languages call "functions" or "procedures") with
: , just add ; before defining the next word with : -- that will help a
lot. At the end, your last word also has to end in ; before you can call it.

BTW, there are many code examples in the test directory; you can look under
test/rp2040, test/rp2350, and test/rp_common in particular.

Travis

amken3d, 2025-02-14:
You are a national treasure Travis. Thank you for your patience and response.

amken3d, 2025-02-14:
I verified my findings again. 
I used these commands one by one 

Welcome to zeptoforth
Built for rp2350, version 1.10.0.1, on Sun Feb 9 04:25:18 PM CST 2025
zeptoforth comes with ABSOLUTELY NO WARRANTY: for details type `license'
 ok
pin import  ok
0 output-pin high 0 pin!  ok
low 0 pin!  ok
high 0 pin!  ok
low 0 pin!  ok
26 output-pin high 26 pin!  ok
low 26 pin!  ok
31 output-pin high 31 pin!  ok
low 31 pin!  ok
32 output-pin high 32 pin!  ok
34 output-pin high 34 pin!  ok
low 3 pin!  ok
low 32 pin!  ok
31 output-pin high 31 pin!  ok
low 31 pin!  ok
low 32 pin!  ok
high 32 pin!  ok
46 output-pin high 46 pin!  ok
high 46 pin!  ok
low 46 pin!  ok

I could see the voltage jump for 0, 26 and 31. For pins 32 and 46 the output remained flat.  What do you think?

tabemann, 2025-02-14:
> I verified my findings again.
> I used these commands one by one
>
> Welcome to zeptoforth
> Built for rp2350, version 1.10.0.1, on Sun Feb 9 04:25:18 PM CST 2025
> zeptoforth comes with ABSOLUTELY NO WARRANTY: for details type `license'
> ok
> pin import ok
> 0 output-pin high 0 pin! ok
> low 0 pin! ok
> high 0 pin! ok
> low 0 pin! ok
> 26 output-pin high 26 pin! ok
> low 26 pin! ok
> 31 output-pin high 31 pin! ok
> low 31 pin! ok
> 32 output-pin high 32 pin! ok
> 34 output-pin high 34 pin! ok
> low 3 pin! ok
> low 32 pin! ok
> 31 output-pin high 31 pin! ok
> low 31 pin! ok
> low 32 pin! ok
> high 32 pin! ok
> 46 output-pin high 46 pin! ok
> high 46 pin! ok
> low 46 pin! ok
>
> I could see the voltage jump for 0, 26 and 31. For pins 32 and 46 the
> output remained flat. What do you think?
>
>
> I verified my findings again.
> I used these commands one by one
>
> Welcome to zeptoforth
> Built for rp2350, version 1.10.0.1, on Sun Feb 9 04:25:18 PM CST 2025
> zeptoforth comes with ABSOLUTELY NO WARRANTY: for details type `license'
> ok
> pin import ok
> 0 output-pin high 0 pin! ok
> low 0 pin! ok
> high 0 pin! ok
> low 0 pin! ok
> 26 output-pin high 26 pin! ok
> low 26 pin! ok
> 31 output-pin high 31 pin! ok
> low 31 pin! ok
> 32 output-pin high 32 pin! ok
> 34 output-pin high 34 pin! ok
> low 3 pin! ok
> low 32 pin! ok
> 31 output-pin high 31 pin! ok
> low 31 pin! ok
> low 32 pin! ok
> high 32 pin! ok
> 46 output-pin high 46 pin! ok
> high 46 pin! ok
> low 46 pin! ok
>
> I could see the voltage jump for 0, 26 and 31. For pins 32 and 46 the
> output remained flat. What do you think?

I figured it out -- there was a bug in OUTPUT-PIN and INPUT-PIN where it
would always access GPIO_OE_SET and GPIO_OE_CLR even if high GPIO's (> 31)
were being controlled. I have made a fix to this and am compiling my fix
right now. I should get the code into the devel and master source trees
today, and put out a bug-fix release over the weekend.

If you feel up to it, and don't want to wait for this weekend, after I
confirm that it builds and all, you can git clone the repo and build
zeptoforth using the bin/1.10.0.1/rp2350/zeptoforth_kernel-1.10.0.1.uf2 as
the basis for a build (i.e. you won't need to install the arm-none-eabi
toolchain yourself) per the instructions in README.md. All you will need is
Python 3 and the ability for pip, as directed by utils/codeload3.sh (which
assumes bash), to install pySerial in a local environment, or if you don't
have bash (e.g. your using BSD or macOS)  you can use pip to install
pySerial manually and then use utils/codeload3.py directly.

Travis

amken3d, 2025-02-14:
I will try it out tomorrow. Thank you for the quick fix

tabemann, 2025-02-15:
>
>> I will try it out tomorrow. Thank you for the quick fix
>>
>>
>> I will try it out tomorrow. Thank you for the quick fix
>>
>>
> I have successfully compiled the changes and pushed them out to devel and
> master. Mind you, I haven't actually tested them because I don't have an
> RP2350 board that actually exposes any of the high GPIO's to the user
> without using a special adapter cable I don't have. However, I am confident
> that the fix will work.

The fix is in zeptoforth release 1.10.1, which can be gotten from
https://github.com/tabemann/zeptoforth/releases/tag/v1.10.1.

Travis

amken3d, 2025-02-15:
Pins are all working as expected. Thanks for adding the terminal colors.

tabemann, 2025-02-15:
> Pins are all working as expected. Thanks for adding the terminal colors.
>
>
> Pins are all working as expected. Thanks for adding the terminal colors.
Thanks for trying out the high GPIO's with 1.10.1. Note that if you have
any RP2040 boards you will want to get zeptoforth 1.10.1.1, which is at
https://github.com/tabemann/zeptoforth/releases/tag/v1.10.1.1, as that
release specifically fixes major bugs in rebooting and the watchdog on the
RP2040 (note that there is no need to install 1.10.1.1 on your RP2350 or
other boards, because only the RP2040 binaries have been updated in
1.10.1.1).

Travis