tabemann/zeptoforth #185: RTC functions not working

intecele, 2025-02-15:
I tried the newest version of zeptoforth (1.10.1.1) on pico 2 and it seems that the rtc functions don´t work. When i use
rtc import
date-time@

the board hangs an there is no return to command line.

Have i done anything wrong ?

Thank you very much Travis for the rest of this excellent work !

Tom

tabemann, 2025-02-15:
> I tried the newest version of zeptoforth (1.10.1.1) on pico 2 and it seems
> that the rtc functions don´t work. When i use
> rtc import
> date-time@
>
> the board hangs an there is no return to command line.
>
> Have i done anything wrong ?
>
> Thank you very much Travis for the rest of this excellent work !
>
> Tom
>
>
> I tried the newest version of zeptoforth (1.10.1.1) on pico 2 and it seems
> that the rtc functions don´t work. When i use
> rtc import
> date-time@
>
> the board hangs an there is no return to command line.
>
> Have i done anything wrong ?
>
> Thank you very much Travis for the rest of this excellent work !
>
> Tom
RTC::DATE-TIME@ expects an argument that is a data structure that it
populates, as in the following:

rtc import
date-time-size buffer: my-date-time
my-date-time date-time@
my-date-time date-time.

which on my Pico 2 W, which I had not set the time on or installed the NTP
client on, displayed:

Thu, 01 Jan 1970 08:39:06 ok

Travis

intecele, 2025-02-15:
Hello Travis,
many thanks for your fast help. Now i can follow the intention of the rtc functions. 
Great ! 
Sorry for my fault.
Have a good time and best regards
Thomas

tabemann, 2025-02-15:
> Hello Travis,
> many thanks for your fast help. Now i can follow the intention of the rtc
> functions.
> Great !
> Sorry for my fault.
> Have a good time and best regards
> Thomas
>
>
> Hello Travis,
> many thanks for your fast help. Now i can follow the intention of the rtc
> functions.
> Great !
> Sorry for my fault.
> Have a good time and best regards
> Thomas
Oh, no need to apologize! Most @ words return value(s) on the data stack,
so that RTC::DATE-TIME@ instead takes the address of a structure it
populates might be unexpected.

Travis