tabemann/zeptoforth #187: key and key?

intecele, 2025-02-19:
Hello all together,
i use key or key? in some words. sometime ist works in other uses alway return 10 (decimal). Has anybody an idea, what i make wrong ? Something left to inialize ? i use zeptoforth-full-usb-1.10.1.1.
thanks a lot
Thomas

tabemann, 2025-02-19:
> Hello all together,
> i use key or key? in some words. sometime ist works in other uses alway
> return 10 (decimal). Has anybody an idea, what i make wrong ? Something
> left to inialize ? i use zeptoforth-full-usb-1.10.1.1.
> thanks a lot
> Thomas
>
>
> Hello all together,
> i use key or key? in some words. sometime ist works in other uses alway
> return 10 (decimal). Has anybody an idea, what i make wrong ? Something
> left to inialize ? i use zeptoforth-full-usb-1.10.1.1.
> thanks a lot
> Thomas
10 (decimal) is linefeed. It seems your terminal emulator is sending CRLF
newlines *to* zeptoforth, and zeptoforth only expects to receive CR
newlines as input, even though like any properly behaved program it sends
CRLF newlines as output to the terminal emulator over USB CDC or serial.
Consequently, the CR is getting caught by zeptoforth and treated as a
newline, while the LF is getting enqueued by the console layer and is
getting output as 10 by KEY .

I can elicit this behavior with the following picocom command, which
replaces output (to USB CDC) CR with CRLF:

$ picocom /dev/ttyACM0 -b 115200 --omap crcrlf

With this configuration, I get the following:

key . 10  ok

from zeptoforth.

Travis

intecele, 2025-02-20:
Hi Travis,
many many thanks. That was the reason. I use 2 different PCs with Teraterm, one with CR the other with CR+LF. Sorry for the inconvenience. 

Have a nice day and all the best 
Thomas

closed 2025-02-20 by intecele