tabemann/zeptoforth #207: Suggestion: merging into Picomite firmware

VOC-LINK, 2025-03-13:
I wonder whether it could be possible to replace BASIC present in Picomite's ( https://geoffg.net/picomite.html ) firmware by Zeptoforth completely, giving us as a result Jupiter Ace of our time — just much better one?

tabemann, 2025-03-13:
zeptoforth could definitely be adapted for the Picomite. Some things would
need to be added, e.g. an HDMI driver would need to be written, and some
things would have to be changed, e.g. the USB CDC console would need to be
replaced with a USB host driver with support for things like keyboards.
Note that this would not be merging zeptoforth into the Picomite firmware
but a wholesale replacement of the Picomite firmware with zeptoforth.

VOC-LINK, 2025-03-13:
What I had in mind was Picomite's firmware already do contain the needed components/drivers — they are handled by one core, that's responsible for picture generation etc. — and although its BASIC implementation can be nice, still I prefer Forth — so it would be a matter „just” to replace that „interpreter part”, handled by the second core. That's why I came up with the mentioned idea.

tabemann, 2025-03-13:
That idea sounds attractive, but the matter is that zeptoforth is an
operating system and expects to be in full control of the underlying
hardware. Furthermore, it does not integrate well with, say, C code because
it is very particular about the handling of particular registers,
specifically R6 (the top of the data stack) and R7 (the address of the
second item on the data stack); in zeptoforth R6 and especially R7 must
always be those things because its interrupt handlers are written mostly in
Forth and thus C code which would disturb those registers cannot be run
unless interrupts are disabled, which has its own problems.

Travis

znmeb, 2025-03-13:
> Furthermore, it does not integrate well with, say, C code because
> it is very particular about the handling of particular registers,
> specifically R6 (the top of the data stack) and R7 (the address of [...]

Hmmm ... I was under the impression that GCC does not clobber r4 - r11 on the RP2040. At least that's what Stephen Smith's Apress book on assembly programming for the RP2040 says. There's a whole section in the book on interfacing assembly and GCC.

Being able to call GCC libraries in the SDK (at least the ones that don't touch things other than registers) would be quite useful. I'd love to be able to use `TinyUSB` UAC2 and MIDI devices from Zeptoforth, for example.

https://www.google.com/books/edition/RP2040_Assembly_Language_Programming/lUyazgEACAAJ?hl=en

tabemann, 2025-03-13:
The only registers one is allowed to *clobber* in the standard ARMv6-M,
ARMv7-M, and ARMv8-M ABI's are R0-R3, but that just means they are caller
rather than callee-saved. The problem here, though, is that if R7 is
modified at all in a fashion different from using it as a stack (which is
different from clobbering it) in foreign code, an interrupt handler written
in Forth which expects R7 to point to the second element of a valid data
stack will misbehave.

Travis