tabemann/zeptoforth #217: Alternate led blinking example

tylerdq, 2025-04-08:
Love this project. Unfortunately the [example code](https://github.com/tabemann/zeptoforth/wiki/Getting-Started-with-zeptoforth) for toggling and blinking the on-board led didn't work on my Raspberry Pi Pico 2 W for some reason. However, I was able to modify the blinker code by using the example in the following GPIO section to blink an LED I wired up as follows:

```
pin import
15 output-pin
: blinker ( -- )
  begin key? not while 15 toggle-pin 500 ms repeat
  key drop
;
blinker
```

Sharing in case this helps anyone else just getting started.

closed 2025-04-08 by tylerdq

tylerdq, 2025-04-09:
I also see that I missed how the introduction explains that the `led` module "specifically does not work on the Raspberry Pi Pico W" for reasons I assume also apply to the Pico 2W.

tabemann, 2025-04-09:
Yes, this is also true about the Pi Pico 2 W -- I just hadn't gotten around
to updating what you refer to. The matter is that with both the Pi Pico W
and Pi Pico 2 W the on-board LED is controlled via GPIO 0 on the CYW43439
chip and not via GPIO 25 on the RP2040 or RP2350 like on the Pi Pico and Pi
Pico 2. You need to install the CYW43439 firmware and the CYW43439 driver
to be able to access the LED on the Pi Pico W and Pi Pico 2 W, and the API
is different as you need to access the LED via the
PICO-W-CYW43-NET-IPV4::<PICO-W-CYW43-NET-IPV4> class in
extra/rp_common/pico_w_cyw43_net_ipv4.fs (or after it is released, also
the PICO-W-CYW43-NET-IPV6::<PICO-W-CYW43-NET-IPV6> in
extra/rp_common/pico_w_cyw43_net_ipv6.fs). Yes, it's a headache,
unfortunately.

Travis