tabemann/zeptoforth #218: Clarifying fat32-tools stack notation

tylerdq, 2025-04-11:
Following [the documentation for fat32-tools](https://github.com/tabemann/zeptoforth/blob/master/docs/words/fat32_tools.md), I can create, list, and change directories and files. But the stack notation for several of these words is in the form `( addr bytes – )` or `( path-addr path-u – )`, which I understand would mean I need to provide 2 values on the stack before executing words like `change-dir`, `list-dir`, or `included`. However, when I simply provide a compiled string like `s" ./"`, `s" examples"`, or `s" example.fs"` before the word I'm executing they all work as expected, leaving no values on the stack afterward. I realized this after supplying 2 values such as `current-fs@ s" examples/example.fs" included` and, upon inspecting the stack with `.s`, finding that not all of the tokens were consumed, leaving some on the stack.

I suppose I'm misreading the stack notation and/or under-informed about how the basic operations work, but after searching the documentation and online resources I couldn't find where to learn more about the relevant fundamentals. Would you be willing to clarify some of the input terms from these examples?

tylerdq, 2025-04-11:
Posting this reminded me to re-read the documentation for `s"` and sure enough it places both of the needed values on the stack as specified by `( compiled: – c-addr u )` in [the documentation](https://github.com/tabemann/zeptoforth/blob/master/docs/words/basic.md#s-2). Apologies for not double-checking that before opening the issue!

closed 2025-04-11 by tylerdq

tabemann, 2025-04-11:
> Posting this reminded me to re-read the documentation for s" and sure
> enough it places both of the needed values on the stack as specified by (
> compiled: – c-addr u ) in the documentation
> Apologies for not double-checking that before opening the issue!
>
>
> Posting this reminded me to re-read the documentation for s" and sure
> enough it places both of the needed values on the stack as specified by (
> compiled: – c-addr u ) in the documentation
> Apologies for not double-checking that before opening the issue!

Oh no problem! Yes, S" pushes both the address of the string's first byte
and its length in bytes on the data stack. Note that when used from within
compiled code the address it pushes is within the compiled code itself, but
when used from interpretation mode it pushes the address of a string in a
'temporary buffer' that is ephemeral in nature (a total of 256 bytes is
alloted for the contents of the temporary buffer, and if the current
pointer into the temporary buffer does not have enough room to allot a
contiguous section of data it will wrap around to the start -- note that
256 bytes is the maximum length of the input buffer anyways so this is
normally not a problem as long as you treat interpreted strings as not
valid beyond the end of an interpreted line of code).

Travis

tylerdq, 2025-04-12:
Got it. Thanks so much for the reply and the additional helpful context! For the record, I think the project documentation is excellent and has helped me get up and running quite quickly with some basic knowledge from _Starting Forth_.