tabemann/zeptoforth #223: cannot download image due to minidict being always empty

nonchip, 2025-04-29:
I set up a rp2350 (pimoroni pico 2 plus w) with a few things loaded (mostly zed, transfer, the sram FS stuff), installed my own cornerstone and saved the minidict.

running `utils/download_uf2_image.sh rp2350 /dev/ttyACM0 custom_16mib` to extract the uf2, outputs the following (removing the TX lines full of forth code):

```
Uploading src/common/forth/ihex.fs
TX:[snip]
grep: screenlog.0: No such file or directory
:00000001FF clone_end 
Uploading src/common/forth/ihex_minidict.fs
TX: [snip]
rm: cannot remove 'screenlog.0': No such file or directory
:00000001FF clone_end 
arm-none-eabi-objcopy: error: the input file 'custom_16mib.minidict.ihex' has no sections
No screen session found.
```
and the minidict ihex file is indeed just `:00000001FF`

nonchip, 2025-04-29:
turns out a sequence of `reset` (my cornerstone, just to make sure), **forcing** the minidict save, and then downloading *without* a reboot did it, i don't understand *why* that worked though :D

tabemann, 2025-04-29:
On Tue, Apr 29, 2025, 07:16 Kyra Zimmer ***@***.***> wrote:

> *nonchip* left a comment (tabemann/zeptoforth#223)
> <https://github.com/tabemann/zeptoforth/issues/223#issuecomment-2838569314>
>
> turns out a sequence of reset (my cornerstone, just to make sure),
> *forcing* the minidict save, and then downloading *without* a reboot did
> it, i don't understand *why* that worked though :D
>

What you observed is very curious -- I will have to look into it further.
One thing that would help me diagnose this issue is if you provided an
exact step-by-step listing of each command you executed (including words at
the zeptoforth REPL and pseudowords handled by utils/codeload3.py (which
utils/codeload3.sh is a thin wrapper around) or zeptocom.js). This would be
extremely helpful in trying to figure out the underlying problem.

I should note that at the end of a normal build on the RP2350 (and RP2040)
code is installed so that on bootup if the flash dictionary has changed the
minidictionary is regenerated and written to flash -- but this should not
affect utils/download_uf2_image.sh at all.

Travis

nonchip, 2025-05-05:
> I should note that at the end of a normal build on the RP2350 (and RP2040)
code is installed so that on bootup if the flash dictionary has changed the
minidictionary is regenerated and written to flash

i mentioned that above too with "though the platform code does include save-minidict so shouldnt even need to" which is why i find it so strange that a forced save was required.

> One thing that would help me diagnose this issue is if you provided an
exact step-by-step listing of each command you executed

that's relatively easy:
- installed rp2350-16mib full-usb
- codeloaded:
  - `extra/rp2350/setup_pico_plus_2_psram_fat32.fs`
  - `extra/common/transfer_all.fs`
  - `extra/common/zeptoed_all.fs`
- manually executed:
  - `compile-to-flash   cornerstone RESET   compile-to-ram`
  - `mini-dict::save-flash-mini-dict`
  - any combination of:
    - nothing
    - `reboot`
    - `RESET`
    - the RST button
    - powercycling
  - optionally: `words` to double check the dict still correctly ends in RESET
- so far so good. everything as expected. downloading now:
  - `utils/download_uf2_image.sh rp2350 /dev/ttyACM0 custom_16mib` errors as above, producing empty dict dump.
- fix:
  - `mini-dict::force-save-flash-mini-dict   reboot`
  - `utils/download_uf2_image.sh rp2350 /dev/ttyACM0 custom_16mib` suddenly works.

tabemann, 2025-05-08:
On Mon, May 5, 2025 at 12:17 AM Kyra Zimmer ***@***.***>
wrote:

> *nonchip* left a comment (tabemann/zeptoforth#223)
> <https://github.com/tabemann/zeptoforth/issues/223#issuecomment-2849938036>
>
> I should note that at the end of a normal build on the RP2350 (and RP2040)
> code is installed so that on bootup if the flash dictionary has changed the
> minidictionary is regenerated and written to flash
>
> i mentioned that above too with "though the platform code does include
> save-minidict so shouldnt even need to" which is why i find it so strange
> that a forced save was required.
>
> One thing that would help me diagnose this issue is if you provided an
> exact step-by-step listing of each command you executed
>
> that's relatively easy:
>
>    - installed rp2350-16mib full-usb
>    - codeloaded:
>       - extra/rp2350/setup_pico_plus_2_psram_fat32.fs
>       - extra/common/transfer_all.fs
>       - extra/common/zeptoed_all.fs
>    - manually executed:
>       - compile-to-flash cornerstone RESET compile-to-ram
>       - mini-dict::save-flash-mini-dict
>       - any combination of:
>          - nothing
>          - reboot
>          - RESET
>          - the RST button
>          - powercycling
>       - optionally: words to double check the dict still correctly ends
>       in RESET
>    - so far so good. everything as expected. downloading now:
>       - utils/download_uf2_image.sh rp2350 /dev/ttyACM0 custom_16mib
>       errors as above, producing empty dict dump.
>    - fix:
>       - mini-dict::force-save-flash-mini-dict reboot
>       - utils/download_uf2_image.sh rp2350 /dev/ttyACM0 custom_16mib
>       suddenly works.
>
>
I executed the following steps:

- I installed the rp2350_16mib full_usb UF2 on my Pimoroni Pico Plus 2
- I codeloaded extra/rp2350/setup_pico_plus_2_psram_fat32.fs
- I connected to my board with picocom and executed 'compile-to-flash',
then disconnected
- I codeloaded extra/common/zeptoed_all.fs and extra/common/transfer_all.fs
- I connected to my board with picocom and pressed control-C,  rebooting it.
- I connected to my board with picocom and executed 'compile-to-flash',
'cornerstone reset', and 'reboot'
- I executed 'utils/download_uf2_image.sh rp2350 /dev/ttyACM1 custom_16mib'
at my bash prompt

It then generated the correct UF2 file without a problem.

Note that there is generally no reason that the user would need to use
MINI-DICT::SAVE-FLASH-MINI-DICT or MINI-DICT::FORCE-SAVE-FLASH-MINI-DICT ─
these really are internal words not meant for public consumption.

Rather, after compiling to flash, before attempting to do anything else
with your board, including generating a UF2 file with
utils/download_uf2_image.sh, simply reboot it. Rebooting automatically
updates the mini-dictionary in flash.

I hope this answers your questions and provides a resolution to this
problem.

Thanks for the feedback, by the way,

Travis

> Message ID: ***@***.***>
>

nonchip, 2025-05-10:
> Note that there is generally no reason that the user would need to use

that is exactly what i'm saying and that's why i made this issue, because it's wrong.

> Rebooting automatically
> updates the mini-dictionary in flash.

again, it literally does not on my pico.

tabemann, 2025-05-10:
On Sat, May 10, 2025, 10:50 Kyra Zimmer ***@***.***> wrote:

> *nonchip* left a comment (tabemann/zeptoforth#223)
> <https://github.com/tabemann/zeptoforth/issues/223#issuecomment-2868986029>
>
> Note that there is generally no reason that the user would need to use
>
> that is exactly what i'm saying and that's why i made this issue, because
> it's wrong.
>
> Rebooting automatically
> updates the mini-dictionary in flash.
>
> again, it literally does not on my pico.
>

Just for the record, did you execute COMPILE-TO-FLASH immediately before
loading extra/common/zeptoed_all.fs and extra/common/transfer_all.fs?

I ask because you didn't say you did in your list of steps, and if you did
not then these would not have been compiled to flash in the first place.

This is to enable these to be compiled to RAM for temporary usage, e.g. one
may not need zeptoed all the time and want to load it on an as-needed basis.

Travis

>

tabemann, 2025-05-12:
On Mon, May 12, 2025 at 12:46 AM Kyra Zimmer ***@***.***>
wrote:

> *nonchip* left a comment (tabemann/zeptoforth#223)
> <https://github.com/tabemann/zeptoforth/issues/223#issuecomment-2870914420>
>
> yes i did, just like i said in my list of steps. are you actually using
> github issues for your github issues or just reading emails with first
> drafts?
>
>
Yes, I do largely rely on email to handle GitHub Issues rather than the web
interface. I cannot always be logged into GitHub under my tabemann account
for Reasons, but I can practically always access my email, which is why.

Travis

> Message ID: ***@***.***>
>

tabemann, 2025-05-14:
On Mon, May 12, 2025 at 8:55 AM Travis Bemann ***@***.***> wrote:

> On Mon, May 12, 2025 at 12:46 AM Kyra Zimmer ***@***.***>
> wrote:
>
> Yes, I do largely rely on email to handle GitHub Issues rather than the
> web interface. I cannot always be logged into GitHub under [...]

To be honest, I don't know what to say. I went back to your GitHub issue
and followed your directions to reproduce the issue almost exactly (with
sole the exception of that I used a Pimoroni Pico Plus 2 rather than a
Pimoroni Pico Plus 2 W; note that I used REBOOT to reboot) and things
worked perfectly. I was even able to reinstall the generated UF2 file on my
Pico Plus 2 and edit a file in the FAT32 filesystem in PSRAM with zeptoed
and then list the file with FAT32-TOOLS::LIST-FILE.

Travis

> Message ID: ***@***.***>
>>
>